arkypita/LaserGRBL #1816: Uninstaller Data

GreenLightning, 2022-04-23:
Hello,

the installer puts `unins000.dat`, `unins000.exe` into `C:\Program Files (x86)`. Can you please change it, so that it puts it in a subfolder, e.g. into `C:\Program Files (x86)\LaserGRBL` as well.

Thank you.

arkypita, 2022-04-23:
LaserGRBL installer is made by a freeware called innosetup. I don't think I can change the location of that files.

GreenLightning, 2022-04-23:
The `setup.iss` contains a line `UninstallFilesDir={pf}` that should control this location.

Please note the documentation for `UninstallFilesDir` says (the documentation is on [this page](https://jrsoftware.org/ishelp/index.php?topic=setupsection) somewhere, but it does not let me link it directly):

> You should not assign a different value here on a new version of an application, or else Setup won't find the uninstall logs from the previous versions and therefore won't be able to append to them.

closed 2022-04-26 by arkypita