arkypita/LaserGRBL #1817: grbl config error

dottenc12, 2022-04-24:
Grbl v1.1h
Wrapper: UsbSerial
Win 10, Home, 10.0.19043.0 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.8.4470.0 built by: NET48REL1LAST_C

TypeOf exception  [System.TimeoutException]
Exception message [Wrong number of config param found! (29/34)]
Exception source  [LaserGRBL], thread [Main Thread]
Exception method  [Void RefreshConfig()]

   ----------- stack trace -----------
   at LaserGRBL.GrblCore.RefreshConfig() in E:\LaserGRBL\LaserGRBL\Core\GrblCore.cs:line 876
   at LaserGRBL.GrblConfig.BtnExport_Click(Object sender, EventArgs e) in E:\LaserGRBL\LaserGRBL\GrblConfig.cs:line 178
   at System.Windows.Forms.Control.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ButtonBase.WndProc(Message& m)
   at System.Windows.Forms.Button.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

dottenc12, 2022-04-24:
Changed x and y axis maximum speed now axis wont move

arkypita, 2022-04-27:
The Grbl config page throw this error when the number of config param found (29) is less then expected (34)
This mean that your controller reply with a wrong list of parameters to "$$" command.

Can you try using the $$ command in "type gcode here" box and see what is going on?

dottenc12, 2022-04-27:
Yes that is what I did and I saw that all of the parameters were all crazy.
I went in and set them all back to default from the book. My machine is now
back in operation.

Thanks for your help

closed 2022-08-16 by arkypita