arkypita/LaserGRBL #1823: New laser configuration to the makerbase mks dlc v2.0

baselbaransi, 2022-05-09:
Hi there, 
I bought a new laser with 3pin cable, and having trouble to make it work with the board i have the makerbase mks dlc v2.0.
This is the board on the laser
[image]
 The seller recommended to connect it like that, but it didn't work, any ideas on how to make it work?
Or maybe need other software for engraving instead of gprl laser.
Thanks in advance

StuartB4, 2022-05-09:
I think it should be like this picture. The 12v + and - pins supply the power and the S and G are the PWM and Ground.
If that does'nt work connect the G and the - of the 12v pins together and the Yellow on the S pin. 

[image]

StuartB4, 2022-05-09:
If you use the pins marked Laser you will have no control over the power to the laser. Thats what the S (PWM) pin does.
But you need a permanent 12v supply, that is what the 12V+/- pins do.

baselbaransi, 2022-05-09:
Thanks dears, i will try this option

solutiongeek2022, 2022-05-24:
This version is instructive. The [portable laser rust removal machine](https://www.laserchina.com/cleaner/portable-cleaning-machine/) is another option.