arkypita/LaserGRBL #1824: Introduce advanced command line features for easier automation

otonribic, 2022-05-09:
For those among us tinkering with automation and autoprocessing, ideally doing the cutting/engraving work instead of the user would make the entire process of using LaserGRBL much more comfortable.

The most straightforward solution (even if not the easiest) that comes to mind is a more complex/advanced set of command line options, allowing adding and appending files and setting parameters.
E.g., adding two files as an example (adding sugarcube.png and then appending sugar2.svg, each with its own parameters), although syntax may vary wildly depending on the implementation:
`LaserGRBL.exe sugarcube.png /lasermax:400 /speed:1200 /linetoline /linpermm:10 sugar2.svg /speed:600 /lasermax:750`
This would allow the user (coming from another application starting LaserGRBL with these parameters) to just essentially press F12, check the contours, and start.

A more hacky method is to allow keyboard shortcuts for everything (Alt+various keys underlined in Windows) to focus various GUI elements. This would allow external apps to more or less reliably simulate keystrokes and do the same thing automatically what user needs to otherwise do manually. (To mention nothing about speeding things up for the power users ;-) )

Much appreciated

arkypita, 2022-05-23:
Hi, thanks for this tip. I find this feature very useful, but the way the LaserGRBL code is done at the moment is not easy to implement. In the LaserGRBL code there is not much separation between UI code and algorithms, so the management of a command line becomes a bit complex.

closed 2022-05-23 by arkypita

HenniePeters, 2024-10-11:
Instead of using the command line, would it be possible to run a script, started from the menu?
Are you running into the same algorithm limitations then, or would that be easier to implement?

otonribic, 2024-10-12:
Just FYI, how I'm currently "circumventing" it is by using mouseclick and
keyboard simulation scripts (PyAutoGUI for Python) to e.g. load 50 or so
files and append them over each other with various settings.

HenniePeters, 2024-10-13:
Nice trick, but how do you know where to "click"? I guess you run laserGRBL maximised, but then does the file open dialog always open at the same location?

otonribic, 2024-10-13:
On the same screen with same window position, yes it does. But for that
very reason I prefer simulating keystrokes which is, of course, much more
straightforward. I use "tab" strokes to move across controls, always in the
same order.