arkypita/LaserGRBL #1860: for svg (and other specific slicing algorithms), S commands are sent alone on individual lines, without M3/M4

moefear85, 2022-07-04:
**Describe the bug**
Usually all power parameters (for example S255) are present at the end of a G0/G1 command. According to specifications, If S parameter is present without G0/G1, it should be preceded by M3/M4. But in specific modes (such as trying to engrave an svg file but also for some other non-standard planning algorithms), lasergrbl begins to output S commands alone on individual lines, for example:

> G0 X100 Y50
> S100
> G1 X123 Y44
> S10
> ....

This however causes marlin firmware to keep generating "unknown command" after each S line, because marlin is expecting either:
> G1 X123 Y44 S247
or
> G1 X48 Y111
> M3 S209
> ...

**To Reproduce**
Steps to reproduce the behavior:
1. Simply try to prepare any svg file

**Expected behavior**
That M3/M4 preceed any sole Sxxx parameter

**Screenshots**
Will upload soon

**Hardware and software configuration (please complete the following information):**
 - Windows 11
 - Ender-3 with Marlin 2.1.x firmware
 - LaserGRBL version: latest

**Additional context**
I'm a laser cnc noob, so sorry if I missed something i should have researched first.

brakthehun, 2022-08-17:
This was a request I made years ago. My issue was the buffer was getting full as I was doing high speed lasering and it was stuttering. In Grbl,, once you put the board in M3 or M4 mode,, it stays there until a change. Once th redundant M4 was removed,, My lasers stopped stuttering.

moefear85, 2022-08-17:
@brakthehun you're confused. I'm talking about (the most recent) marlin firmware. It doesn't work without M3/M4 commands, irrespective at what speed. SVG wasn't even supported back then!

arkypita, 2022-08-17:
G0/G1 are modal commands (as well as M3/M4). According to specifications for modal commands, once they are sent once they stay active, so is no needs to repeat them. Therefore it should be possible to send lines containing only positions or power changes without needs to repeat M or G command.

[image]

[image]

arkypita, 2022-08-17:
What does Marlin firmware developer say about this behavior?

moefear85, 2022-08-17:
Hi,

I don't know, though laser features are still experimental. I think I need try the "full compatibility mode with lasergrbl" feature, which I haven't gotten around to yet. Once I know for sure if it does/not work, I'll respond.

But the main issue remains the fact that lasergrbl outputs gcode differently depending on the imported file extension. Hence I only encounter this for svg, as well as specific slicing algorithms. I haven't returned to this in quite a while, so I don't remember exactly, but a standard slicing process ends up producing code that marlin accepts without any problems, as it then includes the M parameter on every line. Hence it would be awesome if one can configure this, even for svg files.

I would have used lasergrbl as a firmware, but I need to retain 3D printing capabilities on the same machine.

FarFlyer, 2022-10-03:
I have the same problem with Marlin 2.0.9.2 with M4 inline commands.
With a raster image tracing it's fine and the S parameter is at the end of the G1 move command.
But with SVG or raster image vectorized, the S parameter is on a new line and it doesn't work.

Example of raster image tracing:
`G1 X0 Y0 F3000
M4 S0
G1 X9.9 S200
X10 S187
X10.1 S28
G1 X10.1 Y0.1 S0
G1 X10 S28
X9.9 S187
X0 S200
S0
G1 X0 Y0.2 S0
G1 X9.9 S200
X10 S187
X10.1 S28
G1 X10.1 Y0.3 S0`

Example of vector (for cutting):
`M4 S0
F2000
G1 X0 Y0.085
S50
G1 X9.899 Y9.985
S0
G1 X9.956 Y9.985
S50
G1 X10.007 Y9.933
S0
G1 X10.007 Y9.951
S50
G1 X0.056 Y0
S0`

Is it possible to standardize the output (to the first example)?