arkypita/LaserGRBL #1865: Linux version?

kallisti5, 2022-07-14:
**Is your feature request related to a problem? Please describe.**

Why is there no Linux version?    I'm getting a laser plotter and everyone i've spoken to has Windows in a VM to run just LaserGRBL.

**Describe the solution you'd like**
Linux support

**Describe alternatives you've considered**
Paying a Windows license and running a Windows VM.

arkypita, 2022-07-15:
> Why is there no Linux version?

Because the technology used to build the UI is Winform, and so it is not available on Linux.

> Linux support

Read here: https://lasergrbl.com/faq

arkypita, 2022-07-15:
PS: this is not a bug or an issue, please use discussion for generic question like this.

kallisti5, 2022-07-16:
Thanks!   I chose "Feature request"  when I opened this one :-)
Any chances of reworking Winform to something more cross-platform friendly?

I'd help out, but don't know DotNet.. just C, C++, Rust, Ruby, Golang, Python, etc.   Don't see too many projects like this one written in DotNet :laughing:

MarcioFCMartins, 2022-07-24:
Check out the FAQ linked, it links to an issue where people explain how to get it up and running using wine.

That's how I have been using the program and it runs great so far.