arkypita/LaserGRBL #1868: Viewbox directive doesn't wort properly when ymin <> 0

jjcarron, 2022-07-17:
**Describe the bug**
The Viewbox y-min value is not interpreted correctly

**To Reproduce**
1. I generate a simple square view using OpenScad (The bug is not dependent on the programm but only on non zero Viewbox y-min value)

The code is :  projection () cube (10);

The generated file looks like :
[image]

or
[image]

4.  Open it in LaserGRBL
[image]

Instead to be at **-10** the image is at **-27.8** 

There are two problems :

-  Openscad generates Y-Axis in the wrong way. (positive values are interpreted as negative values (Not your problem, but can be solved with the mirror function on openScad The code looks like : projection () mirror([0,1,0]) cube (10); ) and the generated file looks like : 
[image]
or 
[image]

The image is uploaded correctly in LaserGRBL
[image]

- The second problem stays. I move the object 10 in the y axis withe the code :  projection () mirror([0,1,0]) translate([0,**10**,0 ]) cube (10);
The image looks like 
[image]
or
<?xml version="1.0" standalone="no"?>
<!DOCTYPE svg PUBLIC "-//W3C//DTD SVG 1.1//EN" "http://www.w3.org/Graphics/SVG/1.1/DTD/svg11.dtd">
<svg width="10mm" height="10mm" viewBox="0 **10** 10 10" xmlns="http://www.w3.org/2000/svg" version="1.1">
<title>OpenSCAD Model</title>
<path d="
M 10,10 L 0,10 L 0,20 L 10,20 z
" stroke="black" fill="lightgray" stroke-width="0.5"/>
</svg>

in the LaserGRBL it looks like  
[image]

As you see translated on 27.8 instead of 10

**Expected behavior**
The translation should be done according the value in the Viewbox Statement (with the correct units? I work in mm)

**Hardware and software configuration (please complete the following information):**
 - Windows version 10
 - Engraver brand and model Atomstack P7
 - LaserGRBL version v4.8.4

jjcarron, 2022-07-23:
I found the problem in the code at line 205 of "GCodeFromSVG.cs" 
The line

205: tmp.OffsetY = -vbOffY * svgHeightPx / vbHeight + (svgHeightPx* scale);

should be 

205: tmp.OffsetY = -vbOffY * svgHeightPx / vbHeight + svgHeightPx; 

as the svgHeightPx was already in pixels  at this point

arkypita, 2022-08-16:
Merged your fix in master branch, available with latest LaserGRBL version.
Hope it does not add more issues :-)

closed 2022-08-16 by arkypita