arkypita/LaserGRBL #1880: Raspberry pi4

MArTI9999, 2022-07-30:
Hi.
I trying to install aplication on Raspberry Pi4 with Raspbian OS following instruction without snap by bcabral in FAQ but those commads generates errors.
Also tryed by PlayOnLinux but during installation pops up errors that wine stop working and/or couldn't finde driver_c.
So my question is
Can I install GRBL on Raspberry ,I doing something wrong or just it is for linux strictly?
PS. I am green with linux like systems.

Uweklaus, 2022-08-20:
Hi, I want to install this software on a PI, too. Havn't tried yet. Hope there are no serious issues..

arkypita, 2022-08-20:
LaserGRBL is for Windows and use microsoft .NET framework 4. Running on Linux is possibile via wine/playonlinux (there was report of people successfully running it) but it needs some dependency installed and configured to run

I don't know more because I never tested

Uweklaus, 2022-08-20:
Thx for the comment