arkypita/LaserGRBL #1885: FLUIDNC scraper compatibility.

brakthehun, 2022-08-05:
FluidNC reports motion control a little differently through $axes request and $spindle request.
Current issue is the homing cycle / max move speed / limits / spindle power / laser mode uses a $$ request which FluidNC doesn't respond to the same as GRBL. Only returns $10

arkypita, 2022-08-16:
Is this a FluidNC issue or a LaserGRBL issue?

Can you give a more detailed example of the messages sent by FluidNC?
If possible, can you log the whole communication (LaserGRBL, menu "?", "activate extended log"

brakthehun, 2022-08-17:
[image]
[image]
[image]
[image]

brakthehun, 2022-08-17:
[comlog.txt](https://github.com/arkypita/LaserGRBL/files/9356426/comlog.txt)

arkypita, 2022-08-17:
LaserGRBL is compatible with GRBL, and with firmware capable of behaving like GRBL.

Based on what [I read here](http://wiki.fluidnc.com/en/features/grbl_compatibility) FluidNC it has basic compatibility, and as far as I can see from comlog.txt it seems to me that work quite well.

Feature like $axes and $spindle and other things specific to this firmware are not part of the GRBL specification and therefore are not supported in LaserGRBL.

arkypita, 2022-08-17:
Is there anything I did not understand in your request? Can you rephrase it?

brakthehun, 2022-08-19:
I would like the homing button to show up and the speed max to be set.
I guess I'm asking LaserGRBL to support FluidNC

brakthehun, 2022-09-20:
When scraping,, The max speed is set by :
$100
$101
In FluidNC:
$axes/X/max_rate_per_min:
$axes/Y/max_rate_per_min:

Laser power max is stored in:
$laser/speedmap:
instead of
$30

buster3, 2022-10-09:
I implemented the $30 command in FluidNC. Still, the Focus Toggle in LaserGRBL is not working for me. Any hints? Thanks

[image]

arkypita, 2022-10-09:
What version of LaserGRBL are you using?  Look like an old version that does not support formula in custom button.

buster3, 2022-10-09:
I use the latest v4.9.4

buster3, 2023-01-16:
Any hints here? Should the formula be evaluated by fluidnc or by laserGRBL?

arkypita, 2023-01-16:
By LaserGRBL, but LaserGRBL should be able to read $30 value from controller to be able to evaluate formula.
If this is not possible, just edit custom button and put a fixed S value based on your needs.

brakthehun, 2023-01-17:
That's the issue with the laserGRBL scraper. 
With fluidNC you can query $30 but it doesn't return any value.
You would have to query:
$laser/speed_map
And it would return one or two mins and a max. The max would be the $30 value.

buster3, 2023-01-17:
> That's the issue with the laserGRBL scraper. With fluidNC you can query $30 but it doesn't return any value. You would have to query: $laser/speed_map And it would return one or two mins and a max. The max would be the $30 value.

His does not hold any more for the last fluidNC release. See https://github.com/bdring/FluidNC/issues/648

buster3, 2023-10-06:
Hello, i revisit this issue. Can anybody give me a hint where the '$$' command is implemented. Can I simply add $30 as output? Thanks

buster3, 2023-10-06:
I revisited this issue. Can anybody confirm that LaserGRBL uses $$ and expects a output for $30? How should the output format for $$ look like? Thanks

arkypita, 2023-10-06:
LaserGRBL send $$ than expect to receive the "ok" response (acknowledge to $$).
After ok received, all the lines the macine send is queued into a list (until a timeout since the last message received) then this list is parsed to extract config parameters.

a config parameter is something like "$" at the beginning of a line, followed by numbers, by any spaces, by "=" and then everything. this regular expression is used: "^[$](\d+)\s*=(.*)"