arkypita/LaserGRBL #1890: Connection Fail

kmclain78, 2022-08-07:
Tried to connect LaserGRBL to the laser and immediately received a "exception manager" popup stating something went wrong.  This was just after I updated to 4.9.2.0.
Can not reproduce as now I can not even open GRBL without getting the same error and it immediately telling me to abort.
Three screenshots are loaded for your viewing, I realized they are mostly the same so stopped after the 3rd one and aborted the issue.
I am running Windows 10 fully updated, FoxAlien 20W Reizer Engraver, GRBL 4.9.2.0 updated today (8/7/2022)

Thanks for any help that can be given. 

[image]
[image]
[image]

arkypita, 2022-08-07:
Hi, thanks for reporting. It is definitely something related to the changes I made recently for version 4.9.2
I immediately check the code of the procedure in error, and release a correction as soon as possible.

arkypita, 2022-08-07:
Could you please post me last lines of your "sessionlog" file? You can open this file via menu "?" -> "open session log"

arkypita, 2022-08-07:
and Please, also grbl configuration from menu "grbl" -> "grbl configuration" can help me a lot

kmclain78, 2022-08-07:
Let me know if you can read it. I would love to be able to get to the grbl configuration but I'm unable to get past that screen, it forces me to abort and shuts down.
[image]

arkypita, 2022-08-07:
Luckily I have a foxalien controller to test, I have identified the problem which is specific to how that machine behaves.
it is a non-adherence to the specification, so I will have to try to understand how to manage it.
I need a few minutes.

arkypita, 2022-08-07:
As a documentation of the problem, the fact is that the machine responds to the $$ command by sending the parameters with their description added in parentheses. This is incorrect, because the machine should just return the parameter value.

This complicates the reading of the data by the program (in this case LaserGRBL) and makes it more difficult to manage the configuration of string parameters, because it is no longer clear to define what is value and what is description.

[image]

00011	00015557	tx	$$\n
00012	00015561	rx	\r
00013	00015561	rx	 $0=5 (step pulse, usec)\r
00015	00015586	rx	 $1=25 (step idle delay, msec)\r
00016	00015587	rx	 $2=0 (step port invert mask:00000000)\r
00017	00015588	rx	 $3=2 (dir port invert mask:00000010)\r
00018	00015588	rx	 $4=0 (step enable invert, bool)\r
00019	00015589	rx	 $5=0 (limit pins invert, bool)\r
00020	00015589	rx	 $6=0 (probe pin invert, bool)\r
00021	00015590	rx	 $10=3 (status report mask:00000011)\r
00022	00015591	rx	 $11=0.010 (junction deviation, mm)\r
00023	00015592	rx	 $12=0.002 (arc tolerance, mm)\r
00024	00015595	rx	 $13=0 (report inches, bool)\r
00025	00015597	rx	 $20=0 (soft limits, bool)\r
00026	00015600	rx	 $21=1 (hard limits, bool)\r
00027	00015603	rx	 $22=1 (homing cycle, bool)\r
00028	00015607	rx	 $23=3 (homing dir invert mask:00000011)\r
00029	00015610	rx	 $24=100.000 (homing feed, mm/min)\r
00030	00015613	rx	 $25=2000.000 (homing seek, mm/min)\r
00031	00015619	rx	 $26=250 (homing debounce, msec)\r
00032	00015622	rx	 $27=2.000 (homing pull-off, mm)\r
00033	00015622	rx	 $30=1000 (spindle max rpm)\r
00034	00015625	rx	 $31=0 (spindle min rpm)\r
00035	00015628	rx	 $32=1 (laser mode, bool)\r
00036	00015631	rx	 $100=160.000 (x, step/mm)\r
00037	00015635	rx	 $101=160.000 (y, step/mm)\r
00038	00015637	rx	 $102=160.000 (z, step/mm)\r
00039	00015821	tx	[3F]

kmclain78, 2022-08-07:
Ok... So how do I fix it if I can't get into the program?

arkypita, 2022-08-07:
> Ok... So how do I fix it if I can't get into the program?

No, you can't fix.
I have released new version v4.9.3 and now it should work! 

https://github.com/arkypita/LaserGRBL/releases/download/v4.9.3/install.exe

closed 2022-08-07 by arkypita

kmclain78, 2022-08-07:
Unfortunately it did not... I'll check again later, have to find something else to get some things done as I have shows coming up and need my laser working.  Thank you tho!

[image]
[image]

arkypita, 2022-08-07:
I think your is not working because LaserGRBL store in a file the last configuration it has read from the machine, and now that it have read something wrong is stuck.

Please try deleting (or rename) this file: C:\Users\<your user name>\AppData\Roaming\LaserGRBL\LaserGRBL.Settings.bin

arkypita, 2022-08-07:
Delete when LaserGRBL is not running :-)
Please let me know.

kmclain78, 2022-08-07:
Yes, that worked for me.  Thank you.

MARTIN83-DOT, 2022-12-30:
Help several times I try to enter but this page immediately launches me and the only option is to abort and continue What do I do?
[image]

EddyMatel, 2023-11-09:
> Help several times I try to enter but this page immediately launches me and the only option is to abort and continue What do I do? [image]

I have the same problem.

commit c09443a2 (2022-08-07): fix #1890