arkypita/LaserGRBL #1891: help , the program dont open

naomiaaa, 2022-08-09:
LaserGrbl v4.9.3.0
Grbl v1.1 
Wrapper: UsbSerial
Win 10, Home, 10.0.22000.0 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.8.4515.0 built by: NET48REL1LAST_C

TypeOf exception  [System.FormatException]
Exception message [La cadena de entrada no tiene el formato correcto.]
Exception source  [mscorlib], thread [Main Thread]
Exception method  [Void StringToNumber(System.String, System.Globalization.NumberStyles, NumberBuffer ByRef, System.Globalization.NumberFormatInfo, Boolean)]

   ----------- stack trace -----------
   en System.Number.StringToNumber(String str, NumberStyles options, NumberBuffer& number, NumberFormatInfo info, Boolean parseDecimal)
   en System.Number.ParseDecimal(String value, NumberStyles options, NumberFormatInfo numfmt)
   en LaserGRBL.GrblConfST.ReadDecimal(Int32 number, Decimal defval) en E:\LaserGRBL\LaserGRBL\Core\GrblCore.cs:línea 3364
   en LaserGRBL.GrblConfST.get_MaxRateX() en E:\LaserGRBL\LaserGRBL\Core\GrblCore.cs:línea 3328
   en LaserGRBL.JogForm.UpdateFMax_Tick(Object sender, EventArgs e) en E:\LaserGRBL\LaserGRBL\JogForm.cs:línea 95
   en System.Windows.Forms.Timer.OnTick(EventArgs e)
   en System.Windows.Forms.Timer.TimerNativeWindow.WndProc(Message& m)
   en System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

arkypita, 2022-08-09:
what kind of machine?

naomiaaa, 2022-08-09:
I have the atomstack A5 RPO

arkypita, 2022-08-09:
Please try deleting (or rename) this file: C:\Users<your user name>\AppData\Roaming\LaserGRBL\LaserGRBL.Settings.bin as suggested [here](https://github.com/arkypita/LaserGRBL/issues/1890)

naomiaaa, 2022-08-09:
[image]

Can you please tell me which file?

arkypita, 2022-08-09:
No the path is not this one.

 C:\Users\your user name\AppData\Roaming\LaserGRBL\LaserGRBL.Settings.bin

naomiaaa, 2022-08-09:
is there another route? because I can't find it on my PC, PLEASE

arkypita, 2022-08-10:
If you can't find the path, just check "show hidden elements" in your folder browser.

[image]

naomiaaa, 2022-08-10:
Thanks, I already opened the program, but the FRAME, FOCUS AND BLINK light does not turn on, the machine works and records normally, I don't know what could have happened, I hope you can help me

arkypita, 2022-08-10:
LaserGRBL need to read at least one time the configuration from the machine. please open "grbl -> grbl configuration" when the machine is connected and idle, and verify to see all the configuration values read from machine

closed 2022-08-16 by arkypita

MARTIN83-DOT, 2022-12-30:
The program does not allow me to continue, it immediately launches me a page where it gives me the option to approach and continue what do I do? HELP!
[image]