arkypita/LaserGRBL #1893: problem

taghlubkasha, 2022-08-13:
LaserGrbl v4.9.3.0 Grbl v1.1f Wrapper: UsbSerial Win 10, Professional, 10.0.19044.0 (7) CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.8.4515.0 built by: NET48REL1LAST_C  TypeOf exception  [System.OverflowException] Exception message [Value was either too large or too small for an Int32.] Exception source  [mscorlib], thread [Main Thread] Exception method  [Int32 ToInt32(System.Decimal)]     ----------- stack trace -----------    at System.Decimal.ToInt32(Decimal d)    at LaserGRBL.RasterConverter.ConvertSizeAndOptionForm.AssignMinMaxLimit() in E:\LaserGRBL\LaserGRBL\RasterConverter\ConvertSizeAndOptionForm.cs:line 78    at LaserGRBL.RasterConverter.ConvertSizeAndOptionForm..ctor(GrblCore core) in E:\LaserGRBL\LaserGRBL\RasterConverter\ConvertSizeAndOptionForm.cs:line 49    at LaserGRBL.RasterConverter.RasterToLaserForm.BtnCreateClick(Object sender, EventArgs e) in E:\LaserGRBL\LaserGRBL\RasterConverter\RasterToLaserForm.cs:line 179    at System.Windows.Forms.Control.OnClick(EventArgs e)    at System.Windows.Forms.Button.OnClick(EventArgs e)    at System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)    at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)    at System.Windows.Forms.Control.WndProc(Message& m)    at System.Windows.Forms.ButtonBase.WndProc(Message& m)    at System.Windows.Forms.Button.WndProc(Message& m)    at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)
[image]

LANDERO22, 2022-08-13:
tengo el mismo problema ya encontraste alguna solucion

arkypita, 2022-08-14:
Check your grbl configuration (menu grbl, grbl configuration). Probably you have some parameters with out of range or zero value (like machine size)

⁣Ottieni BlueMail per Android ​

taghlubkasha, 2022-08-14:
Everyone says check check and no one says how do I check where do I go what do I do

arkypita, 2022-08-16:
> Everyone says check check and no one says how do I check where do I go what do I do

Check your grbl configuration **(menu grbl, grbl configuration)**. Probably you have **some parameters with out of range or zero value (like machine size)**

arkypita, 2022-08-16:
parameter 130 and 131

taghlubkasha, 2022-08-17:
??????????

alastaira, 2022-08-19:
@taghlubkasha It would be a lot easier to help you if you could be a lot more specific in describing your problem. Please give more detailed information about what you were doing when the error occurred - were you trying to import an image? What image (filetype/size)? What steps had you taken? Have you tried with a different image instead? What are your GBRL parameters shown in the GRBL->GRBL Configuration screen?

taghlubkasha, 2022-08-19:
Hi how are you
I was trying to increase the power of the laser
I use
Laptop HP
Windows 10
64
Laser SCULPFUN S9
LaserGrbl

في جمعة، 19 أغسطس، 2022 في 11:18 ص، كتب alastaira ***@***.***

>:

> @taghlubkasha <https://github.com/taghlubkasha> It would be a lot easier
> to help you if you could be a lot more specific in describing your problem.
> Please give more detailed information about what [...]
-- 
Taghlub kasha

alastaira, 2022-08-19:
You mean you typed a new value in the S-Max field of the Target Image dialog box? And that's what caused the error message to appear? 
[image]

taghlubkasha, 2022-10-11:
What is the solution, can you tell me?