arkypita/LaserGRBL #1896: track and zoom

nicolab67, 2022-08-14:
I would like to ask if it is possible to have a function that "turns off" the display of the track for fast movements in the main window. In addition, a zoom function would also be very useful.

Sono molto entusiasta di questo software e ho già da parte la bottiglia da stappare per quando importerà dxf :D
Utilizzando dithering per la stampa di immagini, però, non è facile avere la percezione effettiva del risultato finale e il motivo principale è il disturbo causato dalla traccia grigia dei movimenti veloci. Poterla eliminare (forse basta cambiarne il colore come quello di sfondo) sarebbe già un bel passo avanti.
La possibilità, poi, di riuscire a zoomare sarebbe di ulteriore aiuto per questa ed altre occasioni.
Infinite grazie per il tuo lavoro!
Ciao
Nicola

arkypita, 2022-08-16:
LaserGRBL graphics draw the preview line by line without using any accelerated graphics library (opengl, directx etc) so it is very slow. It is not possible to implement a zoom without having to redraw everything at each zoom step. 
I know it's a limit, but it won't be easy to add zoom feature in a short time.

I prefer to keep the tracking lines for rapid movements, as they are indispensable when working on vector graphics. And no, it's not easy to leave them for vector graphics and remove them for raster graphics because there is no difference when drawing preview.

About dxf, it's something I have in the pipeline but I have no time to work on it.

Ciao

arkypita, 2022-08-16:
PS. this is not an issue and should be posted under "discussion". I move it.