arkypita/LaserGRBL #1901: About engraving pause

solawc, 2022-08-16:
Hello, lasergrbl is a very good software. I have used it for a long time. I have MKS dlc32 motherboard, other motherboards ported with STM32, and MKS DLC v1.0 (328p), but no matter which one I use, I have the same problem. I use MKS dlc32 as an example. Instead of using MKS code, my dlc32 uses grbl_ Esp32. Therefore, the problem of insufficient performance caused by the LCD interface can be ignored, as can other mainboards. After the lasergrbl slice is used, a single line of S0 will appear, and the mainboard will stop at the parsing of S0. I click the icon of thunder and lightning, and it can be restarted, but it will still stop. If I modify the NC file myself and delete a single line of S0, the entire NC can be executed normally. Or if I cut the picture in the second method, it will not appear a single line of S0, and the engraving can be completed normally. The above-mentioned modifications and tests are carried out using the same motherboard parameters.

Test the version of lasergrbl: 4.9.3

solawc, 2022-08-16:
I hope someone can help me

arkypita, 2022-08-16:
screenshot can help

solawc, 2022-08-16:
> screenshot can help

[image]
like this,

solawc, 2022-08-16:
[klt.zip](https://github.com/arkypita/LaserGRBL/files/9350080/klt.zip)

this is my test file @arkypita

arkypita, 2022-08-16:
A full screen screenshot is better, as we can see the current position of the laser.

arkypita, 2022-08-16:
Each g-code line LaserGRBL send to the machine, he expect an "ok" message back from the machine. As the controller has room to receive more than one message (usually a buffer of 128 or 256 characters) LaserGRBL can send multiple messages one after the other, even if the machine has not yet answered "ok" to those already sent, until the space is full.

When the buffer space fills up he cannot send any more characters, as long as the machine does not reply "ok" to those already received. This "ok" message received makes the difference between the clock and the green tick.

LaserGRBL cannot know to which message a certain "ok" refers and therefore manages them in a "first sent, first received" logic.

It happens that if LaserGRBL doesn't get back some "ok" messages, it starts to believe that the buffer space is not available. 
When these unconfirmed messages accumulate, space runs out, and therefore LaserGRBL cannot send any more messages.

In this situation, if the buffer unavailability persists for more than 10s, LaserGRBL colors the indicator red and allows you to reset its buffer space counter manually to allow you to start sending data again.

**This is how LaserGRBL works, and could not do otherwise.**

Your situation is generally a symptom of a defect in the serial communication, often caused by electrical noise or defects in the cable or USB port, or connectors. In these cases the machine sends the "ok" but these are lost in the communication.

I asked you for a full screen screenshot because in this case the laser would be positioned at the last position you see in the log (X135.417) confirming that the machine has actually already received and processed all the commands sent including your S0.

If, on the other hand, the problem was really the S0 command (which I exclude as there are already several other S0s in the file you attached) then you would find the laser at position X0, the last position before the S0.

solawc, 2022-08-16:
> Each g-code line LaserGRBL send to the machine, he expect an "ok" message back from the machine. As the controller has room to receive more than one message (usually a buffer of 128 or 256 characters) LaserGRBL can send multiple messages one after the other, even if the machine has not yet answered "ok" to those already sent, until the space is full.
> 
> When the buffer space fills up he cannot send any more characters, as long as the machine does not reply "ok" to those already received. This "ok" message received makes the difference between the clock and the green tick.
> 
> LaserGRBL cannot know to which message a certain "ok" refers and therefore manages them in a "first sent, first received" logic.
> 
> It happens that if LaserGRBL doesn't get back some "ok" messages, it starts to believe that the buffer space is not available. When these unconfirmed messages accumulate, space runs out, and therefore LaserGRBL cannot send any more messages.
> 
> In this situation, if the buffer unavailability persists for more than 10s, LaserGRBL colors the indicator red and allows you to reset its buffer space counter manually to allow you to start sending data again.
> 
> **This is how LaserGRBL works, and could not do otherwise.**
> 
> Your situation is generally a symptom of a defect in the serial communication, often caused by electrical noise or defects in the cable or USB port, or connectors. In these cases the machine sends the "ok" but these are lost in the communication.
> 
> I asked you for a full screen screenshot because in this case (missing "ok" caused by electrical noise) the laser would be positioned at the last position you see in the log (X135.417) confirming that the machine has actually already received and processed all the commands sent including your S0.
> 
> If, on the other hand, the problem was really the S0 command (which I exclude as there are already several other S0s in the file you attached) then you would find the laser at position X0, the last position before the S0.

Yes, I understand what you mean. Through online debugging, I removed the operation of receiving a single S0 from the code, and the whole engraving process became very normal. At present, I have fixed my problem by modifying the source code of the motherboard. Thank you very much for your answer and help

closed 2022-08-16 by solawc