arkypita/LaserGRBL #1903: error

d123pay, 2022-08-17:
LaserGrbl v4.9.3.0
Grbl v1.1 
Wrapper: UsbSerial
Win 10, Home, 10.0.22000.0 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.8.4515.0 built by: NET48REL1LAST_C

TypeOf exception  [System.FormatException]
Exception message [Input string was not in a correct format.]
Exception source  [mscorlib], thread [Main Thread]
Exception method  [Void StringToNumber(System.String, System.Globalization.NumberStyles, NumberBuffer ByRef, System.Globalization.NumberFormatInfo, Boolean)]

   ----------- stack trace -----------
   at System.Number.StringToNumber(String str, NumberStyles options, NumberBuffer& number, NumberFormatInfo info, Boolean parseDecimal)
   at System.Number.ParseDecimal(String value, NumberStyles options, NumberFormatInfo numfmt)
   at LaserGRBL.GrblConfST.ReadDecimal(Int32 number, Decimal defval) in E:\LaserGRBL\LaserGRBL\Core\GrblCore.cs:line 3364
   at LaserGRBL.GrblConfST.get_MaxRateX() in E:\LaserGRBL\LaserGRBL\Core\GrblCore.cs:line 3328
   at LaserGRBL.JogForm.UpdateFMax_Tick(Object sender, EventArgs e) in E:\LaserGRBL\LaserGRBL\JogForm.cs:line 95
   at System.Windows.Forms.Timer.OnTick(EventArgs e)
   at System.Windows.Forms.Timer.TimerNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

arkypita, 2022-08-17:
Can you please post a screenshot of your grbl configuration, parameter 110?

[image]

d123pay, 2022-08-17:
I am starting afresh.  As soon as I reinstall and configure I will.   Thank
you

d123pay, 2022-08-17:
Now I can not get past the Opps something went wrong.  I removed and reinstalled, I am going to try again.

d123pay, 2022-08-17:
After four attempts to uninstall and reinstall I can't get past the Opps something went wrong page.  Lasergrbl never opens just the opps page.

arkypita, 2022-08-19:
Fixed in latest version

closed 2022-08-19 by arkypita