arkypita/LaserGRBL #1918: Der Wert für einen Int32 war zu groß oder zu klein.

derReik, 2022-09-07:
LaserGrbl v4.9.4.0
Grbl v1.1h
Wrapper: UsbSerial
Win 10, Home, 10.0.19043.0 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.8.4515.0 built by: NET48REL1LAST_C

TypeOf exception  [System.OverflowException]
Exception message [Der Wert für einen Int32 war zu groß oder zu klein.]
Exception source  [mscorlib], thread [Main Thread]
Exception method  [Int32 ToInt32(System.Decimal)]

   ----------- stack trace -----------
   bei System.Decimal.ToInt32(Decimal d)
   bei LaserGRBL.RasterConverter.ConvertSizeAndOptionForm.AssignMinMaxLimit() in E:\LaserGRBL\LaserGRBL\RasterConverter\ConvertSizeAndOptionForm.cs:Zeile 88.
   bei LaserGRBL.RasterConverter.ConvertSizeAndOptionForm..ctor(GrblCore core) in E:\LaserGRBL\LaserGRBL\RasterConverter\ConvertSizeAndOptionForm.cs:Zeile 49.
   bei LaserGRBL.RasterConverter.RasterToLaserForm.BtnCreateClick(Object sender, EventArgs e) in E:\LaserGRBL\LaserGRBL\RasterConverter\RasterToLaserForm.cs:Zeile 179.
   bei System.Windows.Forms.Control.OnClick(EventArgs e)
   bei System.Windows.Forms.Button.OnClick(EventArgs e)
   bei System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   bei System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   bei System.Windows.Forms.Control.WndProc(Message& m)
   bei System.Windows.Forms.ButtonBase.WndProc(Message& m)
   bei System.Windows.Forms.Button.WndProc(Message& m)
   bei System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

Der Fehler erscheint wenn ich das Rasterbild importieren möchte.
Was ist hier das Problem?

arkypita, 2022-09-07:
You have something wrong in grbl configuration, parameters machine size width and height. Check your grbl configuration.

derReik, 2022-09-07:
Hm, ich habe die Konfiguration nicht angefasst, es ist alles noch wie es war.

[image]

arkypita, 2022-09-07:
[image]

Maximum travel is fine, but $27 and $30 looks corrupted

derReik, 2022-09-07:
Danke, da werde ich morgen gleich mal alle Werte checken und anpassen.
Wie sich das geändert hat, ist mir aber ein Rätsel.

arkypita, 2022-09-07:
Sometimes it happens that it get corrupted by himself. Nothing you did, nothing LaserGrbl did. Simply the controller eeprom get corrupted.
This is not the first time I see this behavior.

derReik, 2022-09-08:
Okay, ich habe es repariert, danke für die Hilfe.

closed 2022-09-08 by derReik