arkypita/LaserGRBL #1925: EMULATED GRBL V1.1# message ?

Gerd12, 2022-09-16:
EMULATED----Does anyone know how to remove or take action to stop this? After downloading latest 4.9.4, when I start the program, the first time, beside the connect button, is a heading indicates this message instead of the normal line for COM & BAUD rate.
I can load images, but the program seems to be frozen. Nothing is selectable. GRBL code config will not accept any changes, nor any other button will accept a click.
Any help would be appreciated

[image]

StuartB4, 2022-09-17:
Settings on the Grbl menu, if you've not already found it.
[image]

Gerd12, 2022-09-17:
Thank you for the quick reply. Everything works well now. Great PR.

StuartB4, 2022-09-17:
Thats good, happy to help. 👍

closed 2022-09-27 by Gerd12