arkypita/LaserGRBL #1929: Problems with LaserGBRL

Unbbekant, 2022-09-28:
I honestly don't know the problem, but I'm posting this message and that's why I'm here.
[image]

Anything I am available. Cheers

andreealupu85, 2022-10-03:
I have the same problem, any idea how can it be fixed?

alastaira, 2022-10-04:
Please describe what you were doing to make the error occur. Loading a file? What file?

andreealupu85, 2022-10-04:
Indeed it was the file… I fixed it! Thanks a lot 🙏

Unbbekant, 2022-10-04:
> Please describe what you were doing to make the error occur. Loading a file? What file?

To be honest, since I bought it it gives me errors, with the material I work with, with the USB ports, the connection is cut more than I would like. Loading vectors, loading jobs already vectorized and that work for me on other machines. I don't know what else to look for.

andreealupu85, 2022-10-04:
I’ m not having this type of issues, yesterday, the error I mentioned was the first time occurred in 2 years since I use it. And I use it only on leather

arkypita, 2022-10-05:
The point of the code where you have this exception is the point that try to keep refreshed the list of existing COM port.
I don't know the cause, it's outside the code I wrote (system library). I can try to trap the error, but I would like to understand the cause before.

MARTIN83-DOT, 2022-12-30:
The program does not allow me to continue, it immediately launches me a page where it gives me the option to approach and continue what do I do? HELP!
[image]

MARTIN83-DOT, 2022-12-30:
The program does not allow me to continue, it immediately launches a page where it gives me the option to exit and continue, what do I do? HELP!
[image]

MARTIN83-DOT, 2022-12-30:
@Arkypita ATUDA!