arkypita/LaserGRBL #1930: issue

AleJan7, 2022-09-29:
LaserGrbl v4.9.4.0
Grbl v1.1 
Wrapper: UsbSerial
Win 10, Professional, 10.0.18363.0 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.8.4341.0 built by: NET48REL1LAST_C

TypeOf exception  [System.IndexOutOfRangeException]
Exception message [Index was outside the bounds of the array.]
Exception source  [LaserGRBL], thread [Main Thread]
Exception method  [System.Drawing.Color GetPixel(Int32, Int32)]

   ----------- stack trace -----------
   at LaserGRBL.RasterConverter.ImageTransform.DirectBitmap.GetPixel(Int32 x, Int32 y) in E:\LaserGRBL\LaserGRBL\RasterConverter\ImageTransform.cs:line 435
   at LaserGRBL.RasterConverter.ImageTransform.Outline(DirectBitmap bmp, Point pt) in E:\LaserGRBL\LaserGRBL\RasterConverter\ImageTransform.cs:line 545
   at LaserGRBL.RasterConverter.ImageTransform.Outliner(Image img, Point location) in E:\LaserGRBL\LaserGRBL\RasterConverter\ImageTransform.cs:line 535
   at LaserGRBL.RasterConverter.ImageProcessor.Outliner(Point location, Size rsize) in E:\LaserGRBL\LaserGRBL\RasterConverter\ImageProcessor.cs:line 366
   at LaserGRBL.RasterConverter.RasterToLaserForm.PbConvertedMouseUp(Object sender, MouseEventArgs e) in E:\LaserGRBL\LaserGRBL\RasterConverter\RasterToLaserForm.cs:line 679
   at System.Windows.Forms.Control.OnMouseUp(MouseEventArgs e)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lpara
[image]
m)

arkypita, 2022-10-05:
@vniacob any Idea?

vniacob, 2022-10-05:
@AleJan7 please post image file as an attachment and describe operation (where you click with outliner), so we can reproduce test case. Thank you

sheets999, 2023-09-06:
Same problem here.  Any updates on a solution?

LaserGrbl v5.4.0
Grbl v1.1
Windows 10 Pro, OS Build 19045.3324

Open D:\Hutch.svg
System.IndexOutOfRangeException: Index was outside the bounds of the array.
at LaserGRBL.SvgConverter.GCodeFromSVG.parsePathCommand(String svgPath) in E:\LaserGRBL\LaserGRBL\SvgConverter\GCodeFromSVG.cs:line 667
at LaserGRBL.SvgConverter.GCodeFromSVG.parsePath(XElement svgCode, Int32 level) in E:\LaserGRBL\LaserGRBL\SvgConverter\GCodeFromSVG.cs:line 638
at LaserGRBL.SvgConverter.GCodeFromSVG.parseGroup(XElement svgCode, Int32 level) in E:\LaserGRBL\LaserGRBL\SvgConverter\GCodeFromSVG.cs:line 268
at LaserGRBL.SvgConverter.GCodeFromSVG.convertSVG(XElement svgCode, GrblCore core) in E:\LaserGRBL\LaserGRBL\SvgConverter\GCodeFromSVG.cs:line 96
at LaserGRBL.GrblFile.LoadImportedSVG(String filename, Boolean append, GrblCore core) in E:\LaserGRBL\LaserGRBL\GrblFile.cs:line 134
at LaserGRBL.SvgConverter.SvgToGCodeForm.CreateAndShowDialog(GrblCore core, String filename, Form parent, Boolean append) in E:\LaserGRBL\LaserGRBL\SvgConverter\ConvertSizeAndOptionForm.cs:line 47
at LaserGRBL.GrblCore.OpenFile(Form parent, String filename, Boolean append) in E:\LaserGRBL\LaserGRBL\Core\GrblCore.cs:line 659

[image]

svenhb, 2023-09-07:
How did you made this SVG? The number format is really unusual, 
No number seperation like ' ', just one number after the next...
I get this messages:
[image]

sheets999, 2023-09-07:
It was made with the newest version of Adobe Illustrator, not sure the actual version right at this time but I can look when I'm back at that computer.  I ended up doing a bit more trial and error with that file, if I remove the 3 profiles with the curved edges it will import fine.  Something with the irregular curves maybe?

svenhb, 2023-09-07:
The proplem is the unusual way to store the coordinates, but is is allowed in SVG:
This string "m327.81283,23.21606l-.00006,211.49998,68.93206.00002" contains the coordinates
```
327.81283 / 23.21606l
-.00006 / 211.49998
68.93206 / .00002
```
Difficult to seperate this string "68.93206.00002" into two numbers by a regex term.

sheets999, 2023-09-07:
I see that now in the SVG.  I have tried about every option from Illustrator to modify the export to something that doesn't include the inseparable strings but have not had any success.  Maybe I will need to try Inkscape instead.  Thanks for the help.

svenhb, 2023-09-08:
Yes, loading your SVG into Inkscape and following this procedure helps: https://alpha.inkscape.org/vectors/www.inkscapeforum.com/viewtopic388d.html?t=11228