arkypita/LaserGRBL #1941: Iritating issue

VukIgor, 2022-10-17:
Hi All ,have diy diode laser ,well tuned ,worked fine on ligtburn and than ...
Today i have downloaded lasergrbl ,fresh version ,setted it up with my MKS DLC32 V2.1 and everything working fine (except y-axis endswitch which is registered but machine runns anyway ruining my belt)  until first job ,ditter image 260x400 .
Job starts but every 10 seconds stops ,buffer overloads,machine stops but software do not enter in error mode ,if I hit hold than resume it works again for 10 seconds like it never stoped ,no missed lines etc.
Have no clue what is happening,any help is wellcome :D

arkypita, 2022-10-17:
Same here: https://github.com/arkypita/LaserGRBL/issues/1901#issuecomment-1216503981

Not a lasergrbl issue, reveals a communication problem with the control board.

arkypita, 2022-10-17:
sorry: not sure why you say buffer overloads but if you hold/resume it start again. a video of the issue could help

VukIgor, 2022-10-17:
Hi ,thanks for so fast answer :)  ,under buffer overloads i meant when you hover mouse over buffer bar in bottom of software it shows 125 of 127 (whatever that is ) ,in short buffer is full ,machine doesnt run into any error and you can hold it and resume it .
Did use lasergrbl last year but in different setup with ramps etc . However ,mks dlc32 v2.1 works better and faster on lasergrbl than on lightburn with same settings for engraving ,and engraves much much better

VukIgor, 2022-10-17:
@arkypita can lasergrbl run via wifi ? DLC32 has that option and maybe would avoiding usb cable solve this issue,however i dont se any option for except connection to machine via ip .

arkypita, 2022-10-18:
> Hi ,thanks for so fast answer :) ,under buffer overloads i meant when you hover mouse over buffer bar in bottom of software it shows 125 of 127 (whatever that is ) ,in short buffer is full ,machine [...]

The buffer bar showing 125/127 tell you that LaserGRBL has sent 125 characters of commands, and did not receive the corresponding "ok" messages of that messages. In this situation there is no space to send other commands (only 2 characters free) so LaserGRBL waits patiently to receive "ok" to the given commands.

But you have to help me understand what is the cause/situation for which these "ok" do not arrive from the controller. You can see these "ok's" as the green check that is to the right of the command. When there is a clock, the ok is missing/pending.

How these pending commands accumulate is very important in understanding the cause of the problem. That's why I'm asking you for a video.

Do pending commands start growing slowly during streaming, or do they pop up all of a sudden at a certain time?
When it shows 125 of 127, the buffer color is green or is red? Is the "lightning" next to the buffer clickable or not?
What does the "Status: " indicator show?

[image]

Picture or video of the situation help a lot.

arkypita, 2022-10-18:
Furthermore you can record the communication in a file, through the menu "?" - "Activate extended log".

Activate extended log before starting the job, so every command sent and every message received is logged to a file. Start the job, wait the issue to happen, do hold/resume or whatever you do to resume and then zip and upload the file here.

Possibly also the session log (menu "?" - "open session log") can be interesting to see.

arkypita, 2022-10-18:
> @arkypita can lasergrbl run via wifi ? DLC32 has that option and maybe would avoiding usb cable solve this issue,however i dont see any option for that except connection to machine via ip .

Yes, LaserGRBL can run via WiFi.

The laser engraver should be connected to the same WiFi of the PC running LaserGRBL and you should know the IP and tcp PORT of the machine, then you can configure LaserGRBL for telnet connection.

Of course the DLC should support streaming via telnet... i don't know if MKS DLC32 V2.1 support it. According to this page
https://forum.lightburnsoftware.com/t/mks-dlc32-esp32-fw-grbl-wifi-connectivity-success/58735/26?page=2 look like it is possible on port 8080 (i have no idea if it's true)

VukIgor, 2022-10-18:
@arkypita  thank you for your time and help ,i did solve the problem ,replaced that ugly blue usb cable (that was my last option and worked ) but before that sucessfuly connected laser to pc on wifi but had same issue like with old cable and even error message that motherboard (mks dlc32 v2.1) does not respond in time and connection is cut off by software

VukIgor, 2022-10-18:
Here is session log file block that is repeating and coresponds with error message that was shown 

18/10/2022 19:25:05.152	SetStatus   	Machine status [Idle]
18/10/2022 19:25:41.155	EnqueueProgram	Push Header
18/10/2022 19:25:41.155	EnqueueProgram	Push File, 41300 lines
18/10/2022 19:25:41.162	EnqueueProgram	Running program, 41300 lines
18/10/2022 19:25:41.196	SetStatus   	Machine status [Run]
18/10/2022 19:27:33.326	Issue detector	StopResponding [5,-1,-1]
18/10/2022 19:27:52.514	CloseCom    	Close MyMachineIP:8080 [CORE]
18/10/2022 19:27:52.515	SetStatus   	Machine status [Disconnected]
18/10/2022 19:28:02.121	SetStatus   	Machine status [Connecting]
18/10/2022 19:28:02.122	OpenCom     	Open MyMachineIP :8080
18/10/2022 19:28:23.195	OpenCom     	Error: A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond MyMachineIP:8080
18/10/2022 19:28:23.195	SetStatus   	Machine status [Disconnected]