arkypita/LaserGRBL #1943: Welcome to LaserGRBL Discussions!

joewarrentn, 2022-10-31:
### Discussed in https://github.com/arkypita/LaserGRBL/discussions/1209

<div type='discussions-op-text'>

<sup>Originally posted by **arkypita** December 16, 2020</sup>
## 👋 Welcome!

If you have any doubts about LaserGRBL or are looking for help to configure and use your engraver, you can search through the opened discussions: often the answers are already there.

If you don't find an answer you can open a new one.
Remember, to get help more easily always describe:
- which engraver do you have
- what are you trying to do
- what attempts have you already made
- at what point do you stop and what message/error do you have

We’re using Discussions as a place to connect with other members of our community. We hope that you:
- Ask questions you’re wondering about.
- Share ideas.
- Engage with other community members.
- Welcome others and are open-minded, remember that this is a community we build together 💪.

</div>

joewarrentn, 2022-10-31:
I'm testing a LE-4040 with a 40w laser using laser grbl software for possible purchase. I tested one image and it seemed skewed so I tested again with a circle image to verify and it is full scale on width but only 50% on height. Do you know what would be causing this skewed image

P3t3P, 2024-05-05:
My Windows 10 pc is not recognizing the LaserGRBL device drivers so I am unable to connect the software. Can anybody assist?