arkypita/LaserGRBL #1966: Continuous Jog With fluidnc

RDS356, 2022-12-05:
Using Continuous Jog with fluidnc , the laserGRBL only run unit table size 200x300mm, but the softlimits works well.
Is it a laserGRBL issue or must need make another configuration ? Or is not compatibile with FluidNc ?

arkypita, 2022-12-06:
Continuous Jog work sending jog command pointing to the extreme margin of the table size, and then aborting jog command when you release the jog button. If I send jog coordinate greatest then table size, the controller refuse to jog because of soft limit. 

Table size is read from the controller by LaserGRBL ($$ parameters 130 and 131)
I don't know how FluidNc report the table size, but if it was not via $130 and $131, LaserGRBL may use some hardcoded default like 200x300.

[image]

RDS356, 2022-12-06:
To report table size per axes is;

arkypita, 2022-12-07:
Soft limit are controlled by firmware.

LaserGRBL cannot read table size in this format, and you may also have some other problem with lasergrbl on other parameters.

I suggest you a workaround (not sure if it works).

- Set LaserGRBL to talk with it's own grbl emulator (menu Grbl, Settings, Protocol, Connection Protocol), connect to emulator and go to edit Grbl Configuration.
- Set the $ parameters $20, $21, $22, $30, $31, $32, $110, $111, $130 and $131 compatibly with your FluidNc configuration, these are the parameters that LaserGRBL needs to know.
- Disconnect and reconnect from the emulator, to make sure that LaserGRBL has re-read the parameters (maybe not necessary, but we do it just to be sure).
- Reconfigure Connection Protocol to "grbl" and connect to your FluidNC machine.

Now the reading the configuration from FluidNC should fail and therefore LaserGRBL should keep the previous configuration (not sure about this).

It's a try, not sure if it works.
Let me know.

arkypita, 2022-12-07:
Please read my first replay that answer your question

RDS356, 2022-12-08:
Ok, but if soft limit (firmware via) will be stop the machine , why the laser GRBL needs read it before jogging ? Did you understand my doubt ?

RDS356, 2022-12-08:
Maybe I did not explain exactly what I want to say, cause my English is not so good. Please be patient.

arkypita, 2022-12-09:
Continuous Jog **work sending jog** command pointing **to the extreme margin of the table size**, and then aborting jog command when you release the jog button. If I send jog coordinate greatest then table size, the controller refuse to jog because of soft limit.

**Table size is read from the controller by LaserGRBL ($$ parameters 130 and 131)**

RDS356, 2022-12-09:
Thank you again.
Do you have plans to make it 100% compatible with fluidNC ?