arkypita/LaserGRBL #1969: Importing a SVG file causes an IndexOutOfRangeException

Eric13246, 2022-12-08:
I encountered a bug where importing a SVG file causes an IndexOutOfRangeException. When that happens, the program does not crash, but nothing gets loaded in the working canvas.

**To Reproduce**
1. Have an SVG file that was exported from Adobe Illustrator 2022
2. Click on File > Open File
3. Go to the location of the selected SVG file and click Open
4. In the Target image window, leave the default settings and click Create!
5. Click on ? > Open session log to see the error's stacktrace

**Expected behavior**
The SVG file should be properly imported in the working canvas of LaserGRBL.

**Hardware and software configuration (please complete the following information):**
 - Windows 11 Pro 22H2 version 22621.819
 - No engraver connected to PC
 - LaserGRBL version 4.9.4

**Additional information**
The SVG file was generated with Adobe Illustrator 2022. I don't know if this could impact the resulting SVG file, but I just wanted to include that info for reference.

Stacktrace found in the logs when importing this SVG file:

    08-12-2022 10:51:25.379	Program     	------- LaserGRBL v4.9.4 [64bit] START -------
    08-12-2022 10:51:47.808	OpenFile    	Open D:\pots-o-gold.svg
    08-12-2022 10:51:49.080	SvgImport   	System.IndexOutOfRangeException: Index was outside the bounds of the array.
    08-12-2022 10:51:49.080	SvgImport   	   at LaserGRBL.SvgConverter.GCodeFromSVG.parsePathCommand(String svgPath) in E:\LaserGRBL\LaserGRBL\SvgConverter\GCodeFromSVG.cs:line 667
    08-12-2022 10:51:49.080	SvgImport   	   at LaserGRBL.SvgConverter.GCodeFromSVG.parsePath(XElement svgCode, Int32 level) in E:\LaserGRBL\LaserGRBL\SvgConverter\GCodeFromSVG.cs:line 638
    08-12-2022 10:51:49.080	SvgImport   	   at LaserGRBL.SvgConverter.GCodeFromSVG.parseGroup(XElement svgCode, Int32 level) in E:\LaserGRBL\LaserGRBL\SvgConverter\GCodeFromSVG.cs:line 268
    08-12-2022 10:51:49.080	SvgImport   	   at LaserGRBL.SvgConverter.GCodeFromSVG.parseGroup(XElement svgCode, Int32 level) in E:\LaserGRBL\LaserGRBL\SvgConverter\GCodeFromSVG.cs:line 259
    08-12-2022 10:51:49.080	SvgImport   	   at LaserGRBL.SvgConverter.GCodeFromSVG.convertSVG(XElement svgCode, GrblCore core) in E:\LaserGRBL\LaserGRBL\SvgConverter\GCodeFromSVG.cs:line 96
    08-12-2022 10:51:49.080	SvgImport   	   at LaserGRBL.GrblFile.LoadImportedSVG(String filename, Boolean append, GrblCore core) in E:\LaserGRBL\LaserGRBL\GrblFile.cs:line 131
    08-12-2022 10:51:49.080	SvgImport   	   at LaserGRBL.SvgConverter.SvgToGCodeForm.CreateAndShowDialog(GrblCore core, String filename, Form parent, Boolean append) in E:\LaserGRBL\LaserGRBL\SvgConverter\ConvertSizeAndOptionForm.cs:line 47
    08-12-2022 10:51:49.080	SvgImport   	   at LaserGRBL.GrblCore.OpenFile(Form parent, String filename, Boolean append) in E:\LaserGRBL\LaserGRBL\Core\GrblCore.cs:line 648

sheets999, 2023-09-06:
Same problem here.  Any updates on a solution/workaround?

LaserGrbl v5.4.0
Grbl v1.1
Windows 10 Pro, OS Build 19045.3324

Open D:\Hutch.svg
System.IndexOutOfRangeException: Index was outside the bounds of the array.
at LaserGRBL.SvgConverter.GCodeFromSVG.parsePathCommand(String svgPath) in E:\LaserGRBL\LaserGRBL\SvgConverter\GCodeFromSVG.cs:line 667
at LaserGRBL.SvgConverter.GCodeFromSVG.parsePath(XElement svgCode, Int32 level) in E:\LaserGRBL\LaserGRBL\SvgConverter\GCodeFromSVG.cs:line 638
at LaserGRBL.SvgConverter.GCodeFromSVG.parseGroup(XElement svgCode, Int32 level) in E:\LaserGRBL\LaserGRBL\SvgConverter\GCodeFromSVG.cs:line 268
at LaserGRBL.SvgConverter.GCodeFromSVG.convertSVG(XElement svgCode, GrblCore core) in E:\LaserGRBL\LaserGRBL\SvgConverter\GCodeFromSVG.cs:line 96
at LaserGRBL.GrblFile.LoadImportedSVG(String filename, Boolean append, GrblCore core) in E:\LaserGRBL\LaserGRBL\GrblFile.cs:line 134
at LaserGRBL.SvgConverter.SvgToGCodeForm.CreateAndShowDialog(GrblCore core, String filename, Form parent, Boolean append) in E:\LaserGRBL\LaserGRBL\SvgConverter\ConvertSizeAndOptionForm.cs:line 47
at LaserGRBL.GrblCore.OpenFile(Form parent, String filename, Boolean append) in E:\LaserGRBL\LaserGRBL\Core\GrblCore.cs:line 659

[image]

arkypita, 2024-04-09:
Hi, just to give you an update...
I have found the issue, and I am working on a solution.
Hard to find because SVG has a lot of variant so I should deeply check that the fix does not cause more issues :-)

commit cd183c24 (2024-04-09): Fix #1969
closed 2024-04-09 by arkypita