arkypita/LaserGRBL #1977: Simple Marlin 3D based feature (eg : Tronxy XY-3 SE) with laser module

ilaks2, 2022-12-18:
*Lasergrbl cannot connect to a tronxy XY-3SE with a laser module, but connects via Marlin mode.   But it still cannot print in this marlin mode.   However, I realized this laser module is just using the cooling fan which means it gets power via marlin M106 and it gets turned off via M107.  It does not support the G1 Snnnn or G0 S0 commands to turn laser on or off.  (I believe M106 Sxxx ;(0-255) works for power level of laser).   

**Describe the solution you'd like**
Suggestion is in Marlin Mode (or a marlin-2 mode), instead of generating 
G0 X* Y* S0
G1 X* Y* Snnnn   (assuming nnnn<=1000)
Generate instead

M107
G4P1
G1 Fnnnn   ; this is the fast travel speed without travel.   Might be set in your config screen.
G0 X* Y*
M106  S((nnnn==0)?0:((nnnn-1)>>2))
G4P1
G1 Fmmm ; this is the slow travel speed where the slower it goes, the more the laser will engrave.
G1 X* Y*

And add M107 on end of print or after abort.

With this change, pretty much any 3D printer that accepts marlin can be modified to add a laser engraver via the extruder fan power supply as long as the power supply can feed the laser module.

I tested this by doing global replace of all G0 X and G1 X  in the gcode output of lasergrbl and it engraves correctly.

chunlinyao, 2023-01-20:
I have a Tronxy XY-3SE. It works with G0 and G1 commands, also S0 and Snnnn commands.
The only problem I have was LaserGRBL ignore `G1` for continuous G1 commands like this:
```
G1 X111 Y111
X111Y155
X200Y200
```
Tronxy XY-3SE can not parse the line without `G1`

ilaks2, 2023-01-28:
I ended up writing my own bitmap and text render to gcode generator.  the gcode file can be sent to the tronxy and an atomstack A5.   

My quick and dirty suggestion would be to edit the gcode output and insert the G1 and needed spaces in all instances by a global replace of X with "G1 X" and then all cases of Y with " Y"  and then another global replace of G1 G1 with just G1 and G0 G1 with just G0.    That should work.   Also if Snnn or Fnnn has no G1, you can also add the same.   Note it's also easier to code a parser.for that output.  

But note that I don't think the X3-SE handles grayscale well 0 the output power don't seem to be linear so I've only gotten it to work with B&W so far.   Also the focus is really difficult to finetune optimally.

One day I'm going to release the bitmap to grbl generator when I'm done.