arkypita/LaserGRBL #2012: What does this mean

dragonflz, 2023-02-02:
Hey 
I am having an issue with lasergrbl. How do I fix this? Thanks

02/02/2023 11:07:43.722	Program     	------- LaserGRBL v4.9.4 [64bit] START -------
02/02/2023 11:07:43.912	Program     	------- LaserGRBL v4.9.4 [64bit] START -------
02/02/2023 11:07:48.076	Update DB   	Material DB Updated to 2/1/2023 [0660983]
02/02/2023 11:07:48.083	Update DB   	Material DB Updated to 2/1/2023 [0660983]
02/02/2023 11:48:27.768	Program     	---------------- PROGRAM STOP -----------------

02/02/2023 11:57:39.335	SetStatus   	Machine status [Connecting]
02/02/2023 11:57:39.336	OpenCom     	Open COM3 @ 115200 baud Ctrl-X (UsbSerial)
02/02/2023 11:57:39.338	SendImmediate	Send Immediate Command [0x18]
02/02/2023 11:57:41.368	SetStatus   	Machine status [Idle]
_02/02/2023 11:57:51.951	Refresh Config	System.TimeoutException: No response received from grbl!
02/02/2023 11:57:51.951	Refresh Config	   at LaserGRBL.GrblCore.RefreshConfig() in E:\LaserGRBL\LaserGRBL\Core\GrblCore.cs:line 1003
02/02/2023 11:59:45.450	Refresh Config	System.TimeoutException: No response received from grbl!
02/02/2023 11:59:45.450	Refresh Config	   at LaserGRBL.GrblCore.RefreshConfig() in E:\LaserGRBL\LaserGRBL\Core\GrblCore.cs:line 1003
02/02/2023 12:00:03.668	Refresh Config	System.TimeoutException: No response received from grbl!
02/02/2023 12:00:03.668	Refresh Config	   at LaserGRBL.GrblCore.RefreshConfig() in E:\LaserGRBL\LaserGRBL\Core\GrblCore.cs:line 1003_
02/02/2023 12:08:52.815	CloseCom    	Close COM3 [USER]
02/02/2023 12:08:52.816	SetStatus   	Machine status [Disconnected]
02/02/2023 12:08:58.940	Program     	---------------- PROGRAM STOP ---

arkypita, 2023-02-02:
Hi 

"Refresh Config System.TimeoutException: No response received from grbl!" mean that LaserGRBL ask the machine to send back its configuration, but the machine does not respond.

Could you please check what the machine send back if you send the command "$$" in the "type gcode here" box, and press enter?

dragonflz, 2023-02-03:
Hey-
When I try to enter $$ nothing happens my screen just rolls.
[image]

I don't know what happened but I can't do anything.

arkypita, 2023-02-03:
All that purple messages in console, come from controller, tell the controller is continuosly reboot.
I think your machine controller have issue. Contact seller/developer of your machine.