arkypita/LaserGRBL #2032: Change drawing color to better follow progress of the process.

NicoPy, 2023-03-07:
In the Preview window, one can see the path the laser follows when the process runs. When the process is running, a cursor shows the position of the laser.

One enhancement would be to change the color of the drawing progressively to show what have already been processed in the current pass.

StuartB4, 2023-03-08:
I think what you mean is already in the development roadmap. 

[image]

NicoPy, 2023-03-12:
It looks like you are right. I wasn't aware of this roadmap.

StuartB4, 2023-03-12:
👍

arkypita, 2023-03-15:
Hello, this improvement has been in the roadmap for a lot of time.

An attempt was made to implement it but the result was not satisfactory because, given the way the software is made, this modification would require excessive CPU and RAM consumption. In fact, the drawing of the preview is time-consuming, and updating it continuously is currently not sustainable by the part of the code that deals with drawing the gcode.