arkypita/LaserGRBL #2062: .nc is a different size

gaoutlaw07, 2023-05-05:
**Describe the bug**
When I save a specific file to an .nc using either Quick Save or Save (Advanced Options), the frame when using .nc file is smaller than it should be.  When I plug the laser into the computer and open the file with LaserGRBL and click frame in LaserGRBL, it is the correct size.  The frame worked correctly on both the laser and LaserGRBL with a 20mm test square.

**Expected behavior**
I would expect it to frame correctly if it uses a .nc file on removable media or if it is plugged into the computer.

**Screenshots**
The below image shows the upper left corner when running frame from the laser.  This was a screenshot during the actual process. (I hope the image works)
[image]

**Hardware and software configuration (please complete the following information):**
 - Windows 10
 - Atomstack P9 M50
 - Grbl version - I'm not sure
 - v4.9.4

**Additional context**
Add any other context about the problem here.

arkypita, 2023-05-06:
Could you share a sample file?

gaoutlaw07, 2023-05-06:
[pigbank_rotated.zip](https://github.com/arkypita/LaserGRBL/files/11412338/pigbank_rotated.zip)

Absolutely, I made a .zip of both the .nc file and the SVG.  

I had the idea last night of looking into the .nc file to find the maximum x and y coordinates.  I may trying scaling up based on the max x and y coordinate and trying again as a workaround.

arkypita, 2023-05-08:
I have tested your file, both the .svg version and the .nc version with gcode. In both cases LaserGRBL compute and execute framing correctly.

If the problem is when using the file from the SD card, using the machine controller to execute the frame, then the problem is in the machine controller, which fails (or has a bug) in calculating the job size.

In this case, please do not report the problem to me, as it is not a LaserGRBL problem, but report to Atomstack support.

gaoutlaw07, 2023-05-21:
I believe the error exists.  This is now happening with a completely unrelated PNG.  It prints the correct size on a paper, but laserGRBL reads it as a much smaller file (less than half the size).
[image]

gaoutlaw07, 2023-05-21:
It appears that every time I open a file, I have to check Autosize in the _Target image_ window, then click the **EXIF** button to get the image to the correct size.  I don't know if that is the correct workflow, but it is the only path that appears to work.