arkypita/LaserGRBL #2094: M3 bad number format: framing laser mode should be the same as import settings M3/M4

goofyseeker311, 2023-06-26:
v.4.9.4 when you import for creality laser falcon engraver (10W) which uses M4, then the framing button gets the following error.

[image]

also why cant I select/copy the log output text or even have the real time log in a text file, anywhere?

thebruce0, 2023-07-01:
I have this same problem, using the same model (CR-Falcon 10W).
Contacted creality support and being passed on to an engineer.
I just updated to 4.9.4, but it also happened for me with v4.8.

arkypita, 2023-07-01:
LaserGRBL is not able to read the machine configuration from controller board. Please open menu grbl - grbl configuration and see what happens

goofyseeker311, 2023-07-01:
yep, if I remember correctly, it cant read the grbl configuration.

thebruce0, 2023-07-01:
I tried using LightBurn and the communication with the Falcon was communicating fine. Seems it might be a bug with LaserGRBL?

goofyseeker311, 2023-07-01:
yep general communication + job starting works, but the grbl parameters fetch and the framing fails

arkypita, 2023-07-01:
S[$30 * 3 / 100] is only a math formula to compute 3% of max power based on board configuration. Simply replace it with S10 (adjust number to have strongest or lightest beam) by editing custom button code. 

To edit, right click over the button and select edit button

goofyseeker311, 2023-07-01:
but why an user needs to do any editing of code? why not fix the actual problem, instead of now having two problems, the original and the user becoming a programmer, for no reason

arkypita, 2023-07-01:
That's easy... because it is not a LaserGRBL problem.

LaserGRBL is written to be used with standard [grbl firmware](a very common standard, with clear specifications, and used by dozens and dozens of different machine manufacturers.): a very common standard, with clear specifications, and used by dozens and dozens of different machine manufacturers. Firmware, is what runs inside the control board, and with which LaserGRBL talks to.

If a manifacturer (creality or any others...) decide to not implement something standard, but instead use something not standard, or they implement it poorly, and without releasing any specs... how can this be solved by me, who don't even have one to test?

What I can do is suggest workarounds, or ask you who report this problem to run some tests for me... But I can certainly cannot correct a problem of others and without having the tools to do so.

arkypita, 2023-07-01:
> I tried using LightBurn and the Falcon seems to be communicating fine

And that's probably because lightburn uses a fixed S-value, instead of trying to calculate it proportional based on the machine configuration, like LaserGRBL does.

goofyseeker311, 2023-07-01:
well, I would use a default value in the lasergrbl software if the actual cant be read from the grbl device.

arkypita, 2023-07-08:
thanks for the hint. if you had time and what to help what you can do is submit your "session log" file (menu ?, open session log) after you have opened the configuration page. In this way I can have the internal error that lasergrbl trap when reading configuration.

a more detailed info would be the using the "com log" feature (menu ?, activate extended log) that records the communication between lasergrbl and engraver, where I can find exactly the request/reply from lasergrbl to machine. You can start lasergrbl, acrivate extended log, connect to the machine, and then go to grbl, grbl configuration page.

goofyseeker311, 2023-07-08:
08/07/2023 18.00.42.990	Program     	------- LaserGRBL v4.9.4 [64bit] START -------
08/07/2023 18.00.45.977	Update DB   	Material DB is up to date
08/07/2023 18.00.48.877	SetStatus   	Machine status [Connecting]
08/07/2023 18.00.48.881	OpenCom     	Open COM4 @ 115200 baud  (UsbSerial)
08/07/2023 18.00.48.951	SendImmediate	Send Immediate Command [0x18]
08/07/2023 18.00.48.962	SetStatus   	Machine status [Idle]
08/07/2023 18.00.48.967	VersionInfo 	Detected Grbl v1.3a
08/07/2023 18.00.50.509	Refresh Config	System.TimeoutException: Wrong number of config param found! (30/34)
08/07/2023 18.00.50.509	Refresh Config	   at LaserGRBL.GrblCore.RefreshConfig() in E:\LaserGRBL\LaserGRBL\Core\GrblCore.cs:line 1003
08/07/2023 18.01.03.459	SetStatus   	Machine status [Home]
08/07/2023 18.01.08.209	SetStatus   	Machine status [Idle]
08/07/2023 18.02.09.663	OpenFile    	Open C:\Users\tapio\Downloads\testnail.jpg
08/07/2023 18.02.49.710	SetStatus   	Machine status [Run]
08/07/2023 18.02.51.459	SetStatus   	Machine status [Idle]

00000	00269445	log	Recording session started @ 08/07/2023 18.05.12
00001	00269507	tx	[3F]
00002	00269757	tx	[3F]
00003	00269759	rx	<Idle|MPos:0.000,0.000,0.000|FS:0,0>\r\r
00004	00270007	tx	[3F]
00005	00270009	rx	<Idle|MPos:0.000,0.000,0.000|FS:0,0>\r\r
00006	00270257	tx	[3F]
00007	00270259	rx	<Idle|MPos:0.000,0.000,0.000|FS:0,0>\r\r
00008	00270507	tx	[3F]
00009	00270509	rx	<Idle|MPos:0.000,0.000,0.000|FS:0,0>\r\r
00010	00270756	tx	[3F]
00011	00270758	rx	<Idle|MPos:0.000,0.000,0.000|FS:0,0>\r\r
00012	00271006	tx	[3F]
00013	00271008	rx	<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>\r\r
00014	00271256	tx	[3F]
00015	00271258	rx	<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100|A:S>\r\r
00016	00271506	tx	[3F]
00017	00271508	rx	<Idle|MPos:0.000,0.000,0.000|FS:0,0>\r\r
00018	00271757	tx	[3F]
00019	00271758	rx	<Idle|MPos:0.000,0.000,0.000|FS:0,0>\r\r
00020	00272006	tx	[3F]
00021	00272008	rx	<Idle|MPos:0.000,0.000,0.000|FS:0,0>\r\r
00022	00272028	tx	G0X100.000Y0.000\n
00023	00272029	rx	ok\r\r
00024	00272256	tx	[3F]
00025	00272258	rx	<Run|MPos:12.788,0.000,0.000|FS:8100,0>\r\r
00026	00272506	tx	[3F]
00027	00272509	rx	<Run|MPos:51.562,0.000,0.000|FS:10000,0>\r\r
00028	00272756	tx	[3F]
00029	00272758	rx	<Run|MPos:89.225,0.000,0.000|FS:4900,0>\r\r
00030	00273006	tx	[3F]
00031	00273008	rx	<Idle|MPos:100.000,0.000,0.000|FS:0,0>\r\r
00032	00273256	tx	[3F]
00033	00273258	rx	<Idle|MPos:100.000,0.000,0.000|FS:0,0>\r\r
00034	00273506	tx	[3F]
00035	00273508	rx	<Idle|MPos:100.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>\r\r
00036	00273756	tx	[3F]
00037	00273758	rx	<Idle|MPos:100.000,0.000,0.000|FS:0,0|Ov:100,100,100|A:S>\r\r
00038	00274006	tx	[3F]
00039	00274008	rx	<Idle|MPos:100.000,0.000,0.000|FS:0,0>\r\r
00040	00274241	tx	G0X100.000Y0.000\n
00041	00274241	tx	M3S[$30*3/100]F1000\n
00042	00274241	tx	G1Y47.400\n
00043	00274241	tx	G1X177.000\n
00044	00274241	tx	G1Y0.000\n
00045	00274241	tx	G1X100.000\n
00046	00274242	tx	M5\n
00047	00274242	tx	G0X0Y0\n
00048	00274243	rx	ok\r\r
00049	00274243	rx	error:2\r\r
00050	00274245	rx	error:22\r\r
00051	00274246	rx	error:22\r\r
00052	00274246	rx	error:22\r\r
00053	00274247	rx	error:22\r\r
00054	00274248	rx	ok\r\r
[... 150 lines cut ...]
00205	00290257	rx	<Idle|MPos:0.000,0.000,0.000|FS:0,0>\r\r
00206	00290506	tx	[3F]
00207	00290508	rx	<Idle|MPos:0.000,0.000,0.000|FS:0,0>\r\r
00208	00290756	tx	[3F]
00209	00290757	rx	<Idle|MPos:0.000,0.000,0.000|FS:0,0>\r\r
00210	00291006	tx	[3F]
00211	00291008	rx	<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>\r\r
00212	00291256	tx	[3F]
00213	00291258	rx	<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100|A:S>\r\r
00214	00291506	tx	[3F]
00215	00291508	rx	<Idle|MPos:0.000,0.000,0.000|FS:0,0>\r\r
00216	00291756	tx	[3F]
00217	00291758	rx	<Idle|MPos:0.000,0.000,0.000|FS:0,0>\r\r
00218	00292006	tx	[3F]
00219	00292008	rx	<Idle|MPos:0.000,0.000,0.000|FS:0,0>\r\r
00220	00292256	tx	[3F]
00221	00292257	rx	<Idle|MPos:0.000,0.000,0.000|FS:0,0>\r\r
00222	00292506	tx	[3F]
00223	00292507	rx	<Idle|MPos:0.000,0.000,0.000|FS:0,0>\r\r
00224	00292756	tx	[3F]
00225	00292757	rx	<Idle|MPos:0.000,0.000,0.000|FS:0,0>\r\r
00226	00293006	tx	[3F]
00227	00293007	rx	<Idle|MPos:0.000,0.000,0.000|FS:0,0>\r\r

button presses (approximate): corner - frame - corner - frame - corner, corner works, frame does not, just moves a first step then back to origin

goofyseeker311, 2023-07-08:
Refresh Config System.TimeoutException: Wrong number of config param found! (30/34), hmh

arkypita, 2023-07-08:
Thank you for the fast reply, and the start-up log (this was the most important one)

As it can be seen in session log, LaserGRBL expect 34 parameters, but machine send only 30. (`Refresh Config System.TimeoutException: Wrong number of config param found! (30/34)`).
A standard grbl firmware has 34 parameters, Falcon use a non standard one. This cause an exception in reading configuration, and lasergrbl don't have all the parameters to correctly operate the machine.

```
00021	00052756	rx	$0=10\r\r
00022	00052756	rx	$1=250\r\r
00023	00052756	rx	$2=0\r\r
00024	00052756	rx	$3=4\r\r
00025	00052757	rx	$4=0\r\r
00026	00052757	rx	$5=0\r\r
00027	00052757	rx	$6=0\r\r
00028	00052757	rx	$10=1\r\r
00029	00052757	rx	$11=0.010\r\r
00030	00052757	rx	$12=0.002\r\r
00031	00052757	rx	$13=0\r\r
00032	00052758	rx	$20=0\r\r
00033	00052758	rx	$21=0\r\r
00034	00052758	rx	$22=1\r\r
00035	00052758	rx	$23=3\r\r
00036	00052758	rx	$24=1500.000\r\r
00037	00052758	rx	$25=4500.000\r\r
00038	00052758	rx	$26=20\r\r
00039	00052758	rx	$27=3.000\r\r
00040	00052758	rx	$30=1000.000\r\r
00041	00052759	rx	$31=0.000\r\r
00042	00052759	rx	$32=1\r\r
00043	00052759	rx	$100=80.000\r\r
00044	00052759	rx	$101=80.000\r\r
00045	00052759	rx	$110=10000.000\r\r
00046	00052759	rx	$111=10000.000\r\r
00047	00052759	rx	$120=500.000\r\r
00048	00052760	rx	$121=500.000\r\r
00049	00052760	rx	$130=400.000\r\r
00050	00052760	rx	$131=415.000\r\r
```

Look like they (Falcon) decide to remove some parameters they not use, for example ones regarding Z axis (102, 112, 122, 132)
Here is the list of all the official grbl parameters: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

By my side this behavior on the part of the manufacturer is wrong, because it moves from what is a standard to put something on the market that is different from how other machines behave, and it does so without even testing the behavior on one of the popular software.

On the other hand, I can make some small modifications to make lasergrbl more "good-mouthed", in order to accept the configuration even if it lacks data.

arkypita, 2023-07-08:
PS. if you want to edit the button to work, using the workaround I suggest you, here is the code you should write;

`M3 S10 F1000`

`M3S[$303/100]F1000` can't work

goofyseeker311, 2023-07-08:
> PS. if you want to edit the button to work, using the workaround I suggest you, here is the code you should write;
> 
> `M3 S10 F1000`
> 
> `M3S[$303/100]F1000` can't work

why would I need to write any code?

arkypita, 2023-07-08:
> > PS. if you want to edit the button to work, using the workaround I suggest you, here is the code you should write;
> > `M3 S10 F1000`
> > `M3S[$303/100]F1000` can't work
> 
> why would I need to write any code? I still recommend the fail-safe mode. proper handling of exceptions. not assuming everything is right, on any machine. good defaults for failing things. graceful fail assumed operation mode.

I see in your comlog the command "M3S[$303/100]F1000\n"
I assume you have edited the default code of framing custom button, but what you wrote is wrong: $303 does not exist

goofyseeker311, 2023-07-08:
> I see in your comlog the command "M3S[$303/100]F1000\n" I assume you have edited the default code of framing custom button, but what you wrote is wrong: $303 does not exist

no, its default program code. I just use the program, I dont program anything or edit anything. not even parameters if I dont have to.

arkypita, 2023-07-08:
> no, its default program code. I just use the program, I dont program anything or edit anything. not even parameters if I dont have to.

My fault, it is likely that the missing parameter $30 turns $30*3 into $303
Ignore my last messages.

goofyseeker311, 2023-07-08:
bug? in the log?

arkypita, 2023-07-08:
no, custom button use math evaluation (they can compute formula, like "width + 5") but in this case the formula $30*3 is missing of the $30 variable, and so turns out in $303 (or something like this)

arkypita, 2023-07-08:
If you want you can try this pre-release and tell me if it works for you: https://github.com/arkypita/LaserGRBL/releases/download/v4.9.5/install.exe

goofyseeker311, 2023-07-08:
> If you want you can try this pre-release and tell me if it works for you: https://github.com/arkypita/LaserGRBL/releases/download/v4.9.5/install.exe

yes it works and grbl configuration parameters show up properly.

btw, the lightburn way of doing the framing is to loop frame until
its stopped. now it returns to the origin after a single frame pass.

thebruce0, 2023-07-08:
> btw, the lightburn way of doing the framing is to loop frame until
> its stopped. now it returns to the origin after a single frame pass.

Actually I haven't been to get lightburn to do more than a 1 frame pass on the latest version I have. So now if you're saying lasergrbl is also doing a 1 frame pass, perhaps there's another command setting that needs to be adjusted which lightburn hasn't done either.

goofyseeker311, 2023-07-08:
> > btw, the lightburn way of doing the framing is to loop frame until
> > its stopped. now it returns to the origin after a single frame pass.
> 
> Actually I haven't been able to get lightburn to do more than a 1 frame pass on the latest version I have. So now if you're saying lasergrbl is also doing a 1 frame pass, perhaps there's another command setting that needs to be adjusted which lightburn hasn't done either.

lightburn does passes until its stopped, by the stop button. Idk what you mean by that.

thebruce0, 2023-07-08:
> lightburn does frame passes until its stopped, by the stop button. Idk what you mean by that. "it" refers to lasergrbl in this case.

I mean in Lightburn, when I hit frame, it only does one pass. With the PC in the next room I don't get there in time to see the framing. I know I can slow it, but the point is it only does one and doesn't continue the framing until stopped.
I have a tablet that I use remote desktop so I can be next to it to see the one framing pass when I hit the button.

So if Lasergrbl is now also doing only 1 pass with this update, perhaps it's the same issue, and both apps need to adjust the command to make it do continued framing until stopped?

That is an assumption based on this comment:
> btw, the lightburn way of doing the framing is to loop frame until 
> its stopped. now it returns to the origin after a single frame pass.

goofyseeker311, 2023-07-08:
> I mean in Lightburn, when I hit frame, it only does one pass. With the PC in the next room I don't get there in time to see the framing. I know I can slow it, but the point is it only does one and [...]

I dont have your issue with lightburn, it loops until stopped. so I dont think its it.

thebruce0, 2023-07-08:
> I dont have your issue with lightburn, it loops until stopped. so I dont think its it.

I know, I'm just saying, in my lightburn, it only does one loop. So perhaps IF lasergrbl is now only doing one loop, it's a related _updated_ command issue.

goofyseeker311, 2023-07-08:
> > I dont have your issue with lightburn, it loops until stopped. so I dont think its it.
> 
> I know, I'm just saying, in my lightburn, it only does one loop. So perhaps IF lasergrbl is now only doing one loop, it's a related _updated_ command issue.

no, I think lasergrbl has always done just one loop in frame. but idk, has it?