arkypita/LaserGRBL #2095: Handle broken OK! BLINK, DIY CO2 50W, GRBL, ARDUINO UNO, CNC SHIELD

bartart90, 2023-07-05:
Hello
I'm just starting fun with laser cutting and building my own CO2 laser. Basically, it is done just having some problems with GRBL.
Every single time when I plug in the power supply for drivers to CNC Shield and after plugging/connecting ARDUINO UNO into my PC my laser just shows "HANDLE BROKEN OK!" also after homing it does show up (but homing does work, the machine is able to home itself).
Anyway, I could use GRBL for some testing looking at how motors work, setting homing, limit switches pretending laser cutting 
(instead of a CO2 laser I had just simple LED light )
Everything was working well, except showing the problem "HANDLE BROKEN OK!"
Now the time just came to plug my CO2 tube and start doing some real cutting and engraving.
And from now it seems like this "HANDLE BROKEN OK!" might be a real issue.
When I try to set mirrors for my CO2 I have to use the button " BLINK" just to make sure my alignment for mirrors is okay. 
First BLINK okay, Second BLINK okay, Third BLINK error, weird errors.
In the pictures, you can see

[image]
[image]
[image]
[image]
[image]

Once I even had an ALARM DOOR. 
Basically, I having a problem when I try to BLINK my TUBE 50W, but I believe that the problem starts with HANDLE BROKEN OK!
Any ideas?

**Hardware and software configuration (please complete the following information):**
 - WIN10
 - DIY CO2 50W
 - 1.1H
 - LaserGRBL version 4.9.4
 
Hope someone might find a solution, I thinking to erase all data from ARDUINO UNO and start all again.

arkypita, 2023-07-06:
Change USB cable with a better shield one.
Https://LaserGRBL.com/faq

bartart90, 2023-07-06:
Thanks. It was in my mind too. I will do it soon and I will let you know.

arkypita, 2023-07-06:
None of this purple message should be seen if communication is fine. They are position and status message, handled internally and normally not shown.

If you see them it's because they are unrecognizable because of wrong characters added by electric noise

arkypita, 2023-07-06:
Same the errors (red dot) shown messages sent from LaserGRBL and not correctly received by the controller

bartart90, 2023-07-06:
So, I just changed cable that connects Arduino uno with the PC and still has the same problem.
When I plug the power supply to the CNC shield and after connecting the USB cable from the Arduino Uno to my PC , the laser head starts moving like it is trying to find the starting point. ( Important note, the LaserGRBL doesn't even connect) 
When I press connect in LaserGRBL Handle Broken OK! Still appears.

bartart90, 2023-07-06:
I just did clear EPROM memory on Arduino Uno and upload GRBL 1.1H again and seems like Handle Broken OK! Has gone. I need to set up all things again and I Will let you know if it's working correctly.

bartart90, 2023-07-07:
Hi
I jus did set up all things. And the Handle Broken OK! Has gone and not showing again.
I also didn't set homing $H for this time because I think I was happening because of homing.
But now when I press BLINK button it does blink but only once.
It is showing commands with clock next to it and buffer getting slowly full when I press BLINK.
[image]

Any ideas why it isn't responding on my commands?

bartart90, 2023-07-07:
I tried to disconnect LaserGRBL and connect but it just doesn't respond on connect button.
Just nothing happens.
Had to unplug GRBL from PC and plug again.
After that I could take only one BLINK and the program shows purple command again[image]

Seems like there is still noise in connection.

alastaira, 2023-07-10:
Those cheap Arduino CNC shields are notoriously poorly designed - they offer very little isolation between the high current 12V supply that the drivers provide to the stepper motors compared to the 5V logic of the Arduino, including the serial interface used by LaserGRBL. Given that you've said you've built the machine yourself, have you done anything to reduce noise on the stepper motor circuits? (i.e. Do you have appropriate diodes in place, etc.)

bartart90, 2023-07-10:
I haven't. This topic getting more often and I know that I have to figure out something.
Maybe you propose something.
capacitors, dioda, shielded wires?

arkypita, 2023-07-10:
First of all, find if the issue is really electrical noise or a defective board.
Disconnect the board from anything else: laser, motor, switches, fan, and turn off fridge, electronic tools, neon tube and anything with motors or coils that can induce electrical noise.

Use only the board connected with USB cable & disconnected from power supply. Board should run grbl firmware and talk with lasergrbl without any issue. If not, the arduino board is a defective one (or the cable).

Test the board with a test job (i.e. an engraving of a big raster image with 10lines/mm that take 10 minutes or more) let it work until the end. You should have no error at the end of file processing. If you have error test again with a different cable.

Then step by step add something: the board power supply, turn on/off some light, connect laser beam... only at the end re-connect motors.

At each step repeat the test job to check at what point the issue return visible.
Now you have identified the source of your problem.

Once you have the cause we can discuss solutions.

BE CARE: always disconnect power when plugging/unplugging motors or you will damage motor driver IC.

bartart90, 2023-07-18:
Hello
Just as an update, I bought a new Arduino Uno (orginal one) i was using cheapest clone before. Also a proper power supply for CNC shield and I changed wiring for laser tube.
And the problem has gone 😁🙈

Thank you for all your help, support and spend time.

arkypita, 2023-07-18:
Umpteenth confirmation of how much Chinese arduino boards suck (not all of them, but they are often the cause of defects).
If you like you can join [lasergrbl facebook community](https://www.facebook.com/groups/lasergrbl) to share your experience with others