arkypita/LaserGRBL #2097: When Linux version

srem1, 2023-07-08:
Please make in repositores Ubuntu and Fedora

kallisti5, 2024-07-31:
Lightburn just announced discontinuing the Linux version.  It would be great if someone got this working on Linux.

Too bad it's .NET though... you won't find many cross-platform .NET developers

wvengen, 2024-12-19:
The [LaserGRBL FAQ](https://lasergrbl.com/faq/) mentions that you can use [PlayOnLinux](https://www.playonlinux.com/) via [this script](https://github.com/arkypita/LaserGRBL/raw/master/POL_LaserGRBL_setup.sh), or Wine + Mono [as explained here](https://github.com/arkypita/LaserGRBL/issues/5#issuecomment-873564055).
The way to run .NET applications as native as possible is really to use [Mono](https://gitlab.winehq.org/mono/mono) ([old website](https://www.mono-project.com/)). There have been .NET programs packaged in Linux distributions, including some using windows forms (e.g. [openmcdf](https://packages.debian.org/bookworm/openmcdf), [keepass2](https://packages.debian.org/bookworm/keepass2), [gendarme](https://packages.debian.org/bookworm/gendarme)). So it looks like it would be possible to package this for Linux distributions. Often, however, this happens by someone else than the original author, so it probably requires someone stepping in to package this.

Also, Linux distributions generally lag in the published version, to provide stability. This is great for servers, but not for users who want the latest thing to be supported. So for end-user-applications like this, I would suggest that perhaps something like a [Flatpak](https://flatpak.org/) could be easier to use (and perhaps also to package). For .NET applications using a Linux-compatible desktop framework, there is actually a [Flatpak guide](https://docs.flatpak.org/en/latest/dotnet.html). But there seem also to be applications on [Flathub](https://flathub.org/) that use Windows Forms, see [this search in the Flathub manifests](https://github.com/search?q=org%3Aflathub+%22Windows.Forms%22&type=code) (e.g. [Fieldworks](https://github.com/flathub/org.sil.FieldWorks), [Kochbuch](https://github.com/flathub/de.flose.Kochbuch)).
Note that there also seem to be wine-based applications packaged on Flathub, see [this code search](https://github.com/search?q=org%3Aflathub%20org.winehq.Wine&type=code) (see also [Flatpak extensions](https://docs.flatpak.org/en/latest/extension.html)), that could also be a route to make this easily installable.

Rage997, 2025-09-22:
Honestly, I think a Flatpak would be the most practical route here. Native packaging for every distro is a lot of overhead, but a Flatpak (or even AppImage) would make it much easier for people on Ubuntu, Fedora, etc. to just install and run without chasing dependencies. Wine/Mono hacks are fine for testing, but having a clean, sandboxed package feels like the way forward if anyone’s up for maintaining it.