arkypita/LaserGRBL #2098: Update .NET Framework target

alastaira, 2023-07-08:
The LaserGRBL C# project currently targets .NET Framework v4.0, which has been out of support since 2016(!)

Furthermore, the current version of Microsoft Visual Studio (2022) no longer includes .NET Framework components for .NET Framework 4.0 - 4.5.1 (see https://learn.microsoft.com/en-us/dotnet/framework/install/guide-for-developers), which means that developers wanting to contribute to the LaserGRBL project cannot do so unless they are willing to install an out-of-date version of Visual Studio.

I suggest updating the target framework to the latest supported version, v4.8.1, which may address some issues of instability users have faced with the current framework, and also will allow other developers to contribute PRs.

arkypita, 2023-07-08:
If I'm correctly informed, the latest framework version supported by windows XP is version 4.0
Although obsolete, XP is still widely used on secondary PCs, often used to run lasers.

alastaira, 2023-07-10:
Hmm, that is a fair point, although unfortunate. I understand why you'd prioritise ensuring the widest possible userbase rather than convenience for developers though.

closed 2023-07-10 by alastaira