arkypita/LaserGRBL #2106: LaserGrbl v5.0.0.0 issue

samaia0, 2023-07-20:
LaserGrbl v5.0.0.0
Grbl v
Wrapper: UsbSerial
Win 7, Professional, 6.1.7601.65536, Service Pack 1 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.8.4069.0 built by: NET48REL1LAST_B

TypeOf exception  [System.ObjectDisposedException]
Exception message [Safe handle has been closed]
Exception source  [mscorlib], thread []
Exception method  [Void DangerousAddRef(Boolean ByRef)]

   ----------- stack trace -----------
   at System.Runtime.InteropServices.SafeHandle.DangerousAddRef(Boolean& success)
   at System.StubHelpers.StubHelpers.SafeHandleAddRef(SafeHandle pHandle, Boolean& success)
   at Microsoft.Win32.UnsafeNativeMethods.GetOverlappedResult(SafeFileHandle hFile, NativeOverlapped* lpOverlapped, Int32& lpNumberOfBytesTransferred, Boolean bWait)
   at System.IO.Ports.SerialStream.EventLoopRunner.WaitForCommEvent()
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.ThreadHelper.ThreadStart()

samaia0, 2023-07-20:
May it be due to the length of USB cable?

arkypita, 2023-07-20:
It's an error that came from the os, not in my code, never understand cause. i suggest install/update microsoft .NET framework 4
Also can help: In LaserGRBL, settings, switch connection protocol from USBSerial to USBSerial2

samaia0, 2023-07-20:
Thanks!
Not because of too long USB cable?

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Nie
zawiera wirusów.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

ORTUR-laser-master3, 2023-12-26:
Проблема с LaserGrbl v5.4.0
У меня нетбук самсунг с виндовс 7 
Купил ORTUR laser master 3
Проблема в том что программа LaserGrbl v5.4.0 не видит оборудование (гравер ORTUR laser master 3) при нажатии кнопки подключиться.
В диспечере устройств на нетбуке кликая правой кнопкой на Гравер - Пишет - Оборудование работает исправно
В чем проблема? 
Совместима работа программы LaserGrbl v5.4.0 с виндовс 7 и ORTUR laser master 3

ORTUR-laser-master3, 2023-12-26:
проблем на другом компьютере нет, там виндовс 10