arkypita/LaserGRBL #2111: possible bug: cutting line errors on 5.0.0

goofyseeker311, 2023-07-26:
v5.0.0 vectored path mode shows correct tracing image, but results in partially failed cut.

[image]

Do you think the lasergrbl or creality falcon 10W at 200mm/min causes the problem?

arkypita, 2023-07-26:
what is the original image?

goofyseeker311, 2023-07-27:
[image]
[dsl18b-v2h2a-2v.nc.zip](https://github.com/arkypita/LaserGRBL/files/12179780/dsl18b-v2h2a-2v.nc.zip)
resulting gcode for reference if there is a bug.

arkypita, 2023-07-27:
There is no bug, you are using it wrong.

Line2line is for engraving grayscale images in raster mode: use it for pictures, not cut.
Vectorization is the right tools, but your image should by draw as a silhouette. Your images has double edges, any raster line (as thin as you want) has 2 edges, and LaserGRBL vectorization find them both (unless you use "centerline" vectorization).

Better prepare your image like this:
[image]

Understand the concept?

goofyseeker311, 2023-07-27:
well what about the extra lines that was the original issue? and the source image is a CAD software image (or pdf vector source), so its supposed to be lines and not filled rectangles.

goofyseeker311, 2023-07-27:
I stress, the gcode that gets quicksaved from that source image (vectorized) adds some lines that are not in the source image, producing finger cuts in the otherwise rectangle sections.

goofyseeker311, 2023-07-27:
as a side note, pressing hold then stop will not completely stop the project from running.

goofyseeker311, 2023-07-27:
also the center line mode fails to find some center lines on the x0 side of the source image files, and somehow the X-coordinate of all generated files is X0.1 and not X0, might be related issue.

arkypita, 2023-07-27:
> well what about the extra lines that was the original issue?

They are not caused by LaserGRBL. If they were, you'd see them in the preview. If you don't believe you can load the exported gcode with a nice third party software like CAMOTICS https://camotics.org that can render a live preview of the job.

I run your file with my laser engraver, here the result.

[image]

I am 99.999% that the issue is hardware side, very high probability your machine has hit a curb and lost its position, or the engine loses its steps in a certain position. If you keep an eye on the machine while it is working you will see that this is the case.

> and the source image is a CAD software image (or pdf vector source), so its supposed to be lines and not filled rectangles. if you need to do by-hand edits to a cad image it will stop/hinder the work flow totally.

Again you are using it wrong. If the source is a vector then save it as .svg (vector format recognized by LaserGRBL) and this will load it as vector. Any other transition from vector to raster (.jpg, .bmp, .png) and then re-vectorized leads to a loss of precision.

If you need to convert a vector file to svg you can use the free inkscape software.

> centerline vectorization? was not clear that it was vectorization, but something else. should be an option inside the vectorization, not its own option. simplify is better always.

These are two completely different algorithms, not an option of either one. They have different configuration parameters and different options: for this reason they deserve two different entries.

> as a side note, pressing hold then stop will not completely stop the project from running.

The machine has communication buffers, a certain number of commands are sent to the machine and are already in the machine control board waiting to be executed. The "abort" button stops sending new commands, but there is nothing you can do about those already sent. The only way to stop immediately that I know of is the reset (lightning bolt) button. You may or may not like it but either way it depends on the firmware, so not on LaserGRBL.

> and somehow the X-coordinate of all generated files is X0.1 and not X0, might be related issue.

Again... vectorizing a raster image produced from a vector file will never give back the original vector.
If you need CAD quality you should not degrade your file rasterizing it. Just keep it as vector.

goofyseeker311, 2023-08-08:
I will change every G0 command to G1 slow movement command, so that the fast G0 movement will not misalign the laser engraver, yep S0 before every G1 when using those as movement only.

arkypita, 2023-08-08:
> I will change every G0 command to G1 slow movement command, so that the fast G0 movement will not misalign the laser engraver, yep S0 before every G1 when using those as movement only.

I think that you can do something better. Edit your machine configuration, via menu "grbl", "grbl configuration" and set lower acceleration. Do an export before change any values, so you can restore it in case of problems.

> edit: you dont seem to support vector pdf data sources. lightburn does.

Lasergrbl is free, lightburn is not.
Supporting .dxf, .dwg, .pdf require coding, that require times, that I don't have.
 
Hope you understand.

goofyseeker311, 2023-08-09:
dont worry, I have plenty, closing to infinity