arkypita/LaserGRBL #2114: Multiple Pass Per-Path Mode

Rybec, 2023-07-30:
**Is your feature request related to a problem? Please describe.**
Sometimes when I'm cutting 3mm birch ply, my laser fails to cut all the way through, even in 4 or 5 passes, and other times it will cut cleanly in only 3 passes.  Now yes, there are cases where this is due to an extra dense patch of core material, but often it fails to cut well when there's nothing wrong with the wood.  This happens especially frequently when I do cuts that have a ton of detailed parts (which I do frequently), where a single pass can take 45 minutes or more.

Today, I wasted almost a whole sheet of 10"x10" birch ply because of this issue.  I've been blaming the problem on humidity, which is higher today, so I thought through the problem, and I realized that if instead of doing each pass over the whole thing at once I did 4 passes on one path, then the next, and so on, maybe it would word better, because the wood has far less time between passes for each cut to cool down and absorb more moisture from the air.  As I am writing this, I'm halfway through a cut testing this theory, and the results are absolutely awesome.  Despite the conditions being about the same as they were when I wasted 90% of a sheet of wood earlier today, _every single piece_ of my project is cutting all the way through, often within only 3 passes, and far more cleanly.  To achieve this though, I had to copy each path 4 times in my SVG file and then group them to guarantee they would cut in direct succession.

**Describe the solution you'd like**
I would like to have an alternative multi-pass mode, that will do the number of specified passes on _each path_ before moving on to the next, instead of going over the whole program start-to-finish on each pass.  Maybe have a checkbox between the progress bar and the selector for number of passes, that is unchecked by default and when it is checked, it will cut the specified number of passes on each path before moving to the next.

(If LaserGRBL generates the program when loading the file, this might not be feasible.  In that case, maybe have a "Passes per Path" option in the SVG loading dialog, and then generate the GRBL code and add as many copies as needed to the program to get the desired number of passes.)

**Describe alternatives you've considered**
The only viable alternative right now is what I've already done: Duplicate each path 3 more times in Inkscape (which can be done very quickly by selecting everything and duplicating 3 times), and then select all copies of each path and group them together.  This is tedious and time consuming, and it ends up making the number of passes hardcoded into the file, making it poorly suited to distribution to others who may have different laser cutters.

**Additional context**
You wouldn't believe how incredibly nice my cuts are now!  You should honestly make a video suggesting that when people need multiple passes to cut through a material, the passes are done per-path rather than doing the whole thing once on each pass.  Like, this makes such a big difference that it could easily make the difference between being able to cut all the way through a material and not even with a ton of passes.  Even on the parts that did cut properly on the run that failed had far worse edges.  The difference is absolutely huge.  I'm mainly cutting out mechanical stuff, where there tend to be a lot of small parts.  This feature would make an enormous difference for my workflow.  I cannot stress enough how huge of a difference this would make!

Oh right, one more thing: During my current run, I've had my laser get disconnected twice.  I don't know why, and it's not relevant to this.  My cutter doesn't have limit switches, so I have to "rehome" manually, by moving the laser to a corner and then jogging it a set distance to get to the right home location.  This tends to end up with the laser being around half a millimeter off from it's original location, which normally destroys the cut entirely.  Because doing passes on a per-path basis though, I've been able to resume from where it left off, and only the part it was on when it disconnected is affected, because the passes on everything else are finished.  Again, I've wasted whole sheets of wood when disconnects have happened in the past, but I just had two disconnects and only wasted one piece (the second one I was lucky and got it lined up well enough not to destroy it).  This is also absolutely awesome, because it means I didn't waste whole sheets of wood or the two hours that have already gone into this cut.  A feature that made this really easy to achieve would be so _incredibly_ awesome!

Lastly, if you managed to make this far, thank you so much both for making LaserGRBL and for reading all of the above.  I really appreciate your work, and I hope to have the opportunity to enjoy and appreciate it even more when this feature is added.

arkypita, 2023-07-30:
I read your post with interest and I thank you for the suggestion and for the tests you did.

As the management of multiple passes currently works, it is unfortunately not possible to implement what you suggest. If in the future there will be a general redesign I will also include this feature, which seems very interesting to me. For now, I'm putting it in the [roadmap](https://github.com/arkypita/LaserGRBL/issues/64).

As for the problem of disconnections, these are always due to hardware, wiring, or electrical noise.

Rybec, 2023-07-30:
Thanks for the consideration!  I might be able to write a program that will go through the paths and create and group duplicates, and either way, it's not _horribly_ tedious to do it manually (though it did take a good 10 minutes so to do with yesterday's project), and it's certainly worth the savings in time and lost materials.  (I only had two parts cut badly at the end of it.  One was due to the interruption, probably caused by very brief power outage.  I should probably put my cutter on a battery backup.  The other I don't know what happened, but one part was slightly scorched in one corner, but it's still usable, and I had a couple extra of that part on the cut.)

Anyhow, I've done some work on automated SVG generation in the past, using Python.  If I can work out how to parse out the individual paths and make groups of duplicates, I'll write a Python program for doing this and drop it on Github under an open source license.

Rybec, 2023-07-31:
I did it: [https://github.com/Rybec/SVG-Path-Multiplier](https://github.com/Rybec/SVG-Path-Multiplier)

It's not very well tested.  I'll do that some next week and update the documentation.  The SVG file ends up looking about the same (some different path element IDs, but that shouldn't make a difference) as the Inkscape file when doing it manually in Inkscape.  It turns out it was pretty easy.  The hardest part was learning to manipulate XML in Python, which is not too difficult!