arkypita/LaserGRBL #2116: Minor Translation Errors

ristraus, 2023-08-03:
Minor Translation Errors, when LaserGRBL is not loading (if available) a translation.

Main Screen:
Estimated Time:  NOW (now never get translated)

Grbl - Grbl Configuration: All Parameters, Units and Descriptions are always shown in English (Although I should agree that show them in English is safer rather then an incorrect translation)

Grbl - Settings - Protocol: "Emulator", "Buffered" and "Fast" are not translated.

? - License - Is always shown in English.

Empty .resx Files :

\JogForm
\SafetyCountdown
\SplashScreenForm
\Autotrace\AutotraceTest
\Firmware\FlashGrbl
\GrblEmulator\EmulatorUI
\SvgConerter\SvgModeForm
\UserControls\ColorSlider
\UserControls\CommandLog
\UserControls\GrblPanel
\UserControls\LabelTB
\WiFiConfigurator\OrturWiFiConfig

 - Windows version  10
 - Engraver brand and model RSM 40W
 - Grbl version 
 - LaserGRBL version 5.0.1

PS: I submited a pool request with complementary .resx files I forgot to revise when I sent PT_BR translation files 2 weeks ago.

arkypita, 2023-08-03:
Thanks
Most of them are not errors at all. Form or control where there is no text to translate or form (like the license) where to have a single language was a desiderata.

ristraus, 2023-08-06:
Understood.

closed 2023-08-06 by ristraus