arkypita/LaserGRBL #2117: LaserGRBL silently fails to import certain raster images at higher resolution

cbiffle, 2023-08-03:
**Describe the bug**
LaserGRBL (including 5.0) claims to import the attached PNG successfully at 300dpi and resolutions above 4lines/mm, but creates an empty project. If saved, it produces empty gcode.

If you import it using autosize and force DPI to 600dpi (so, half as large on each axis), the import succeeds. Reducing the resolution to 1line/mm also succeeds. (This is with Horizontal tracing. Vertical fails in a slightly different weird way.)

Is there some implicit limit or integer range issue here? It might be worth noting that I'm on a 32-bit platform (WINE, specifically). Nothing is written to the logs when this happens.

**To Reproduce**
Steps to reproduce the behavior:
1. Open the attached PNG file.
2. Configure for Line to Line Tracing, Horizontal, 8 line/mm
3. Click Next
4. Check the Autosize box, which sets to 300 dpi for a 304x67 mm output image.
5. Click Create.

**Expected behavior**
LaserGRBL generates a path for engraving the image, or at least an error message?

**Hardware and software configuration (please complete the following information):**
- WINE 8.10 (configured for 32-bit)
- LaserGRBL 5.0.1 (bug also occurs in the 4.x versions I've tested)

[image]

arkypita, 2023-08-03:
[image]

[image]

Processing take 2 seconds on my PC to generate gcode.

[image]

arkypita, 2023-08-03:
I was able to generate up to 2000x400 that is 40 times bigger (it takes about a minute)

[image]

arkypita, 2023-08-03:
> It might be worth noting that I'm on a 32-bit platform (WINE, specifically)

Yes, the fact that it is 32bit imposes limits on the process especially on memory usage

cbiffle, 2023-08-03:
You're saying that on 64-bit Windows it's working fine?

I can see about reconfiguring WINE, I didn't realize you had a 64-bit release.

cbiffle, 2023-08-04:
...where _is_ your 64-bit release? The installer executable is 32-bit and appears to install 32-bit binaries:

```
LaserGRBL.exe: PE32 executable (GUI) Intel 80386 Mono/.Net assembly, for MS Windows, 3 sections
```

FWIW, installing the 32-bit release in a 64-bit WINE environment does mostly-kinda-work but does not change the behavior when attempting to import raster images with many lines.

Would it be possible to at least get an error message in response to whatever resource allocation failure is happening under the hood? Claiming success and silently creating a size-zero project is confusing, particularly since some other bug causes it to not always update the display when this happens.

arkypita, 2023-08-05:
Executable is a Microsoft .NET Framework executable built for both x86/x64.
The OS decide if to run it in x86 or x64 environment.

I believe you see PE32 executable because the "bootloader" is always 32bit (unless the executable can't run under win32 at all) but the OS switch to x64 just after loading the initial session of the exe.

Keep in mind that .NET is an interpretated language, so it is not rally an executable with compiled code, but an assembly with an intermediate language on it, that the interpreter translate "on the fly" to machine code.

https://en.wikipedia.org/wiki/Common_Intermediate_Language

Actually LaserGRBL trap all its own unexpected exception to a cute form that show all the details, including stack-frame and loaded assembly. If it does not open it should be because this crash is so low-level that my code cannot see it.
I suggest you to get a look in wine/mono logs.

Here, attacched, a version build to be x64 only. Hope this can help your system to load it as x64.
Simply replace the exe

[LaserGRBL.zip](https://github.com/arkypita/LaserGRBL/files/12266271/LaserGRBL.zip)