arkypita/LaserGRBL #2119: Problem: unexecteted reser of board

mdarpamex, 2023-08-06:
Hi everyone,
i try to build a co2 laser machine. "Work correctly" but when send first S command over 25 ($31=0 $30=1000) the job frezee and appears a UnexepetedReset error.
If laser stay in 0-15 range i have no problem, axes move correctly and job terminate successfully

Hardware:
board is ELEGOO UNO R3, with 16u2 chip. Its supply is from usb cable, not from dc cable
laser is 50W co2 laser with 80w power supply. Max 26kV, 28mA

grbl version is the last avaiable
i tryied with laser grbl 5 and the previous one
use windows10 from two different laptop

I tryied shield board from noise with metal grid and aluminium foil around board, general power supply, stepper drivers and usb serial cable. I tested continuity and it works

I have a digital thermometer for water circuit, when laser shot over S30 has some problem like from elettrical noise.

[image]
i draw only X axis but of corse there is Y too.
driver are settet with 6400step/rev with means config $100=80, $101=53.333 step/mm

[image]
[image]

I finish ideas to fix them, can anyone help me?

arkypita, 2023-08-06:
Detach CO2 tube and its driver board from elegoo controller and test it without laser.
Does it does the same issue?

mdarpamex, 2023-08-06:
no, if i disconnect pin11 from board all movement is correct and job finish without problem

arkypita, 2023-08-06:
To me this says two things.

First hypothesis: your laser consumes too much current and the power supply can't handle it (it has a drop in voltage that resets the board) but I would exclude it because if I understand correctly you have separate power supplies.

Second hypothesis: the laser causes electrical disturbances that make you reset the board. I see you use an Arduino UNO clone, some of them are not so good in handling electromagnetic disturbs. Many people have solved all their problems by simply buying an original arduino board.

arkypita, 2023-08-06:
I suggest also an opto-insulator on pin 11

arkypita, 2023-08-08:
Let me know further updates to your issue.

mdarpamex, 2023-08-09:
i bought an optoinsulator and seems to be working properly, but its output courrent is too low and when is connectet to controller laser pin voltate drop from 5v to 1.2v. 
I waiting a new component (a simply level shifter) to have a correct input voltage on laser. 
As soon as it arrives and I test it, I'll write you if it worked. Meanwhile I thank you for the suggestions, it seems to be going in the right direction

mdarpamex, 2023-08-11:
With level shifter not working, but i build this circuit and works:

[image]

I need duplicate schema because this circuit invert signal

closed 2023-08-11 by mdarpamex

arkypita, 2023-08-11:
That's great!