arkypita/LaserGRBL #2122: Laser Power not adjustable

nolden-eab, 2023-08-08:
since Versions higher 5.0.0
it is not possible to adjust the Laserpower for new imports
(Additional fail by exporting files (in .nc files every power values set to 0))

**To Reproduce**
Steps to reproduce the behavior:
1. on Win10 64bit 22H2
without a connected machine
2. Install the newest Version
3. Import open a file (PNG or SVG) 
4. change to M4 dynamic Power
5. try to adjust Laser Power direct or with material DB

**Expected behavior**
1000 = 100%
500 = 50%
1 = 1%

**Screenshots**
[image]
[image]
[image]

**software configuration:**
 - Windows version = Win10 Pro 64bit 22H2
 - Engraver brand and model = NONE
 - Grbl version = NONE
 - LaserGRBL version (5.3.0)

**Additional context**
On Laptop with a Machine not reproducible.
On a new VM reproducible Win10 64bit 22H2.
Reinstalling LaserGRBL does not help. 
Additional deleting C:\Users\...\AppData\Roaming\LaserGRBL before new istall does not help.

arkypita, 2023-08-08:
This issue was reported by few users but no-one was able to provide additional information that points me in the right direction. I hope you can help me solve this issue.

I ask you to collect a log of the communication between LaserGRBL and machine controller

Open LaserGRBL
Go to menu "?" click "activate extended log"
Choose a location where to save the file
Now connect to laser (menu "grbl" click "connect")
Wait for the bottom/right status indicator show "idle"
Go to menu "grbl", "grbl configuration"
Now press "Read"
Close grbl configuration window
Disconnect
Close LaserGRBL

arkypita, 2023-08-08:
Ps: you can always write S1000, S500 etc even if the input is red

StuartB4, 2023-08-08:
No problems here with latest version. Tried line to line, Dithering, and Vectorize. 
Connected and not connected.
Changed the S-MAX setting manually and with the material DB
So I don't think it's a LaserGRBL problem.

Not connected.
[image]

Connected.
[image]

nolden-eab, 2023-08-08:
OK, little misunderstanding.
Problem only exist on Win10 with a fresh installed LaserGRBL WITHOUT a machine ever connected to it. 
(to create files for "offline" mode) (or for software testing)

arkypita, 2023-08-08:
Thanks for your feedback

> with a fresh installed LaserGRBL WITHOUT a machine ever connected to it

This explain all. A fresh install, never connected to a machine, does not know anything about grbl configuration.

In one of the most recent changes (between v4.9.4 of August 2022 and v4.9.5 of a few weeks ago) I tightened the validity check of the read config data, implementing default values (in case the data does not exist) and minimum and maximum values that act as a limit if the controller configuration reports illogical data, to prevent exception and unexpected software behaviour.

[image]

One of these "strengthenings" acts as the default value for SMax ($30) in case the datum has not been read by the machine, and in this case the default datum today is "1". In the previous version the default was 1000.

[image]

This acts as a security element, since I don't know what value the machine has inside and I can't go overboard (if the machine had $30 = 255 and I worked assuming that the value is 1000 I would produce a GCode all at maximum power).

This is probably what is causing problems for some people, it would be necessary to understand why you are not being able to read the configuration in these cases.

In your case it is simple, but many other cases are not explained.
LaserGRBL asks for configuration as soon as the machine is connected, so it should always know the latest configuration.

> (to create files for "offline" mode) (or for software testing)

I suggest to configure lasergrbl to connect to its own emulator, so you can connect to a virtual firmware and you can read/write config, and LaserGRBL can work as if it is connected to a real machine.

[image]

nolden-eab, 2023-08-09:
> I suggest to configure lasergrbl to connect to its own emulator, so you can connect to a virtual firmware and you can read/write config, and LaserGRBL can work as if it is connected to a real machine.
> 
> [image]

sorry but that does not work either. Still the same:
[image]
But im happy that we comming closer to the problem.

Maybe a warning message would be helpful, for new users. Like: you open data but we cant find a connetion or emulator (check connectionprotocoll in settings).

arkypita, 2023-08-09:
> sorry but that does not work either. Still the same:

Look like you have choose to use emulated grbl, but you have not clicked on "connect". So how LaserGRBL can read the values from it?

nolden-eab, 2023-08-11:
Thank you very much now everything is working fine.

closed 2023-08-11 by nolden-eab