arkypita/LaserGRBL #2125: Imported GCode is displayed wrong

CristyRO9, 2023-08-18:
Hi! I have a problem I have a DXF drawing and I converted it to GCode using DXF2GCODE, the problem arises when uploading the GCode to LaserGrbl, the GCode is shown wrong, in the DXF2GCODE preview it's displayed correctly but in LaserGrbl is wrong.What could cause this issue?
[image]
[image]

arkypita, 2023-08-22:
could you attach generated gcode? .zip it and drag drop in github post.

CristyRO9, 2023-08-23:
I'm afraid I don't have it, I deleted it and modified the DXF.