arkypita/LaserGRBL #2130: Install without admin rights

neilferreri, 2023-09-10:
Can I install LaserGRBL without admin rights?
If not, do you have instructions to build from the source?
Thanks!  I'm a teacher trying to help out a student who has a small grbl controlled laser.  We can't install in the school.

arkypita, 2023-09-10:
As far as I know, no admin right are needed to install and run LaserGRBL, but during installation some file extension are associated  to LaserGRBL, and this may write some info in registry, so windows ask for privilege. If you want to build just install visual studio community edition and compile LaserGRBL.

neilferreri, 2023-09-10:
Yes, Windows asks for admin credentials on install.  I'd try to build myself (no real experience there), but installing Visual Studio and all relevant dependencies takes admin access for more than file extension association.
If there's no way to build a version without that credential requirement, I'll look for something else.
Thanks anyway.

arkypita, 2023-09-11:
You don't have any single computer, maybe your computer, where you have admin rights?

arkypita, 2023-09-11:
Here is a .zip file of the latest version. Just unzip and place in the folder you want.
Microsoft .NET Framework is required to run LaserGRBL, hope it is installed.

[LaserGRBL.zip](https://github.com/arkypita/LaserGRBL/files/12572597/LaserGRBL.zip)

closed 2023-09-11 by arkypita

neilferreri, 2023-09-18:
Wow! Perfect! Thank you so much!
I'm sorry for the late reply...I thought I had responded already.

positron96, 2025-01-12:
Hi and thank you for the software!
Would it be possible to provide the portable installation (e.g. such ZIP archives) for releases? I remember they were available a long time ago and I used those to manually upgrade the files in the directory (I have 4.6 on my PC).