arkypita/LaserGRBL #2138: Axis Direction Is Reversed

lawrencek9303, 2023-10-08:
I have two CNC's. 3040ProVerV2 LightBurn, LaserGRBL, Candle and UGS.
When I co connect LaserGRBL it runs the axes the wrong way. The others are correct.
When I ran LaserGRBL on a new GRBL board on a Sainsmart 3018, I had to change the settings in its GRBL.
My Prover V2 grbl is

$0=10
$1=25
$2=0
$3=1
$4=0
$5=0
$6=0
$10=3
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=0
$24=100.000
$25=1000.000
$26=250
$27=3.000
$30=10000
$31=0
$32=0
$100=400.000
$101=400.000
$102=400.000
$110=5000.000
$111=5000.000
$112=5000.000
$120=300.000
$121=300.000
$122=300.000
$130=400.000
$131=300.000
$132=100.000

Thank you

Lawrence

lawrencek9303, 2023-10-17:
Any news on this comment - Please!

arkypita, 2023-10-17:
LaserGRBL use cartesian axis convention, x grow from left to right, y grow from near to far. If your laser react diversely, you can simply adjust grbl axis configuration

lawrencek9303, 2023-10-18:
Hi there,

I cant change the GRBL, because the machine will not run properly with other SW, like CANDLE. Difficult also  to get homing then.
Can I change this in LaserGRBL?