arkypita/LaserGRBL #2139: SVG not correctly loaded

Knut-HH, 2023-10-10:
Just wanted to cut this logo: https://www.maerklin.de/typo3conf/ext/torr_template/Resources/Public/img/logo.svg

It loads correct in Inkscape, for example. But in LaserGRBL 5.4.0 the "m" in the logo is somewhat garbled.
[image]

Knut-HH, 2024-01-24:
[image]
Attached file isn't loaded at all.
Created with inkscape, neither "Inkscape-SVG" nor "Plain SVG" is loaded by LaserGRBL.
Inkscape loads this file just fine.

svenhb, 2024-01-24:
This block is not working, because it is not correct SVG (line 119 - 132) if you remove it, it should work:
(path elements doesn't have width and height)
```
    <path
       style="fill:none;stroke:#000000;stroke-width:0.971339"
       id="rect48"
       height="285.69827"
       x="5.5408549"
       y="787.72485"
       width="0" />
    <path
       style="fill:none;stroke:#000000;stroke-width:0.971339"
       id="rect3"
       width="314.12787"
       x="78.495445"
       y="199.47078"
       height="0" />
```