arkypita/LaserGRBL #2168: Unexpected reset with Arduino UNO clone with CH340G

Fliptronics, 2023-12-18:
In addition to Topic #378 here the solutions of the unexpected resets this board. I buy the Arduino UNO with CH340G for making a laser Engraver and get the UnexpectedReset after a few line of lasering. I look to the board and the schematics and discovered two major errors in the design. 
First problem are the isolation resistors in the Rxd and TxD lines. As you could notice one led is buring less bright then the other. The 1K resistor in the TxD line is placed wrong. When the CH340G pulls the TxD line low, there is a voltage divider with R3, Led1 and R5. The Vf of the red led is around 2V, which means that over the 1K resistors the remaining 3V (5V-2V) is divided, which will be 1.5V over each resistor. This means that this signal is on the ATmega328 input is 1.5V. In the datasheet of the ATmega328 can be seen that the maximum low level is 0.3*Vcc is 1.5 volt. So this is on the edge and with some tolerance in the Vf of the led, it can be worser or a little better. This can be the coarse of the many communication errors. I inspect this with my oscilloscope and indeed it was 1.5 V above ground. Because I don't use the external Rxd or Txd on the header pins, I removed R8 on the board and solder a bridge on the pads and the levels were within specification. When you use external RxD & TxD communication the R8 resistor should be placed between led cathode and ATmega328 Rxd Input. Despite this change the UnExpected reset still took place. This solve the issue of communication errors.
The second mistake on this board is the missing of a decoupling capacitor (100nF) near the atmega328 VCC (pin 4) and ground (pin5) pins. I normally use also a ferrite bead in series and in case of a microcontroller at least a decoupling capacitor at every supply pin is a must for stable operation. At each logic transition a current spike occurs and the energy should be delivered by the local capacitor. 
After I added this capacitor, no UnexpectedResets occur after two runs of 30 minutes, where normally within 2 minutes the reset occurs!

[image]

[image]

arkypita, 2023-12-20:
Hi @Fliptronics 
Totally appreciate your contribution.

martinbystricky, 2024-01-04:
Hi @Fliptronics 
thank you very much for your post. It helped me a lot and hopefully it will help others with this problem.
I add capacitor 100 nF to pin 4 and GND, as you indicated in the picture.

My solution"
[image]

Fliptronics, 2024-01-04:
Hi @martinbystricky  
nice implementation. A 0603 size 100nF, 16V or 25V ceramic capacitor or smaller size will do the job.