arkypita/LaserGRBL #2171: My laser doesn't cut well

Elii3, 2023-12-26:
good morning, I have a sculpfun s9 and I would like to cut some 4 mm thick wood, I set in the materials database the thickness 4mm for the wood and it did not cut, then I tried to put the thickness to 12mm and now it's cutting But I still need to apply a little bit of force to the piece of wood in order to break it off from all the other wood.Does someone knows how to solve this?
[image]
[image]

arkypita, 2023-12-26:
Did you focus  the laser beam properly?
How many passes did you do?

Elii3, 2023-12-26:
> Did you focus  the laser beam properly?
> 
> How many passes did you do?

Yes, I did focus on the lase beam properly and I did 1 pass

Elii3, 2023-12-28:
> Did you focus  the laser beam properly?
> 
> How many passes did you do?

Yes, I did focus the laser beam properly and I did 1 pass

Elii3, 2024-01-27:
> Did you focus  the laser beam properly?
> 
> How many passes did you do?

So, do you know how to solve this?