arkypita/LaserGRBL #2179: 5.7.0 cannot connect

phischy, 2024-01-13:
**Describe the bug**
Laser GRBL will try for a long time to connect and then give a generic "Error connecting to your machine" message. The com port shows up when the machine is plugged into the USB port. Reinstalling the CH340 driver does not fix the problem. Restarting LaserGRBL and the machine also fail to make a difference. The machine was connecting without issue using the previous version.

**Hardware and software configuration (please complete the following information):**
 - Windows version 10 home edition 64-bit
 - Engraver brand and model - Comgrow Z1
 - Grbl version (i.e. 0.9j, 1.1f) 
 - LaserGRBL version 5.7.0

arkypita, 2024-01-13:
Hi, thank you for your bug report.
Please open LaserGRBL, go to menu "?" And click on "show session log".
Scroll down to the last line and see if there are any error message (copy and paste all the file content here,  so I can read it too).

It is useful also to record a full com log (menu ?, enable extended log) and then try to connect to the engraver

phischy, 2024-01-14:
Here is the log. The only error I am seeing is connection timeout.

12/01/2024 18:03:40.753	Program     	------- LaserGRBL v5.4.0 [64bit] START -------
12/01/2024 18:03:40.744	Program     	------- LaserGRBL v5.4.0 [64bit] START -------
12/01/2024 18:03:46.446	Update DB   	Material DB is up to date
12/01/2024 18:04:06.027	Program     	---------------- PROGRAM STOP -----------------

12/01/2024 18:04:32.155	Program     	---------------- PROGRAM STOP -----------------

12/01/2024 18:04:40.594	Program     	------- LaserGRBL v5.7.0 [64bit] START -------
12/01/2024 18:04:44.223	Update DB   	Material DB is up to date
12/01/2024 18:04:47.404	SetStatus   	Machine status [Connecting]
12/01/2024 18:04:47.446	OpenCom     	Open COM8 @ 115200 baud  (UsbSerial)
12/01/2024 18:04:47.466	SendImmediate	Send Immediate Command [0x18]
12/01/2024 18:04:53.791	CloseCom    	Close COM8 [USER]
12/01/2024 18:04:53.795	SetStatus   	Machine status [Disconnected]
12/01/2024 18:05:02.507	SetStatus   	Machine status [Connecting]
12/01/2024 18:05:02.508	OpenCom     	Open COM7 @ 115200 baud  (UsbSerial)
12/01/2024 18:05:07.696	SendImmediate	Send Immediate Command [0x18]
12/01/2024 18:05:07.698	SetStatus   	Machine status [Disconnected]
12/01/2024 18:05:09.951	SetStatus   	Machine status [Connecting]
12/01/2024 18:05:09.952	OpenCom     	Open COM8 @ 115200 baud  (UsbSerial)
12/01/2024 18:05:09.953	SendImmediate	Send Immediate Command [0x18]
12/01/2024 18:05:11.007	CloseCom    	Close COM8 [USER]
12/01/2024 18:05:11.008	SetStatus   	Machine status [Disconnected]
12/01/2024 18:05:28.215	SetStatus   	Machine status [Connecting]
12/01/2024 18:05:28.215	OpenCom     	Open COM8 @ 115200 baud  (UsbSerial)
12/01/2024 18:05:28.216	SendImmediate	Send Immediate Command [0x18]
12/01/2024 18:05:36.170	SetStatus   	Machine status [Disconnected]
12/01/2024 18:05:45.447	SetStatus   	Machine status [Connecting]
12/01/2024 18:05:45.448	OpenCom     	Open COM8 @ 115200 baud  (UsbSerial)
12/01/2024 18:05:45.450	SendImmediate	Send Immediate Command [0x18]
12/01/2024 18:05:55.448	OpenCom     	Connection timeout!
12/01/2024 18:05:55.448	CloseCom    	Close COM8 [CORE]
12/01/2024 18:05:55.451	SetStatus   	Machine status [Disconnected]
12/01/2024 18:06:56.370	SetStatus   	Machine status [Connecting]
12/01/2024 18:06:56.371	OpenCom     	Open COM8 @ 115200 baud  (UsbSerial)
12/01/2024 18:06:56.374	SendImmediate	Send Immediate Command [0x18]
12/01/2024 18:06:59.689	CloseCom    	Close COM8 [USER]
12/01/2024 18:06:59.689	SetStatus   	Machine status [Disconnected]
12/01/2024 18:07:52.605	Program     	---------------- PROGRAM STOP -----------------

12/01/2024 18:07:57.663	Program     	------- LaserGRBL v5.7.0 [64bit] START -------
12/01/2024 18:08:03.395	SetStatus   	Machine status [Connecting]
12/01/2024 18:08:03.402	OpenCom     	Open COM8 @ 115200 baud  (UsbSerial)
12/01/2024 18:08:03.407	SendImmediate	Send Immediate Command [0x18]
12/01/2024 18:08:13.399	OpenCom     	Connection timeout!
12/01/2024 18:08:13.400	CloseCom    	Close COM8 [CORE]
12/01/2024 18:08:13.408	SetStatus   	Machine status [Disconnected]
12/01/2024 18:08:43.842	SetStatus   	Machine status [Connecting]
12/01/2024 18:08:43.843	OpenCom     	Open COM8 @ 57600 baud  (UsbSerial)
12/01/2024 18:08:43.846	SendImmediate	Send Immediate Command [0x18]
12/01/2024 18:08:49.459	CloseCom    	Close COM8 [USER]
12/01/2024 18:08:49.460	SetStatus   	Machine status [Disconnected]
12/01/2024 18:08:54.982	SetStatus   	Machine status [Connecting]
12/01/2024 18:08:54.983	OpenCom     	Open COM3 @ 115200 baud  (UsbSerial)
12/01/2024 18:08:54.984	SendImmediate	Send Immediate Command [0x18]
12/01/2024 18:08:56.338	SendImmediate	System.TimeoutException: The write timed out.
12/01/2024 18:08:56.338	SendImmediate	   at System.IO.Ports.SerialStream.Write(Byte[] array, Int32 offset, Int32 count, Int32 timeout)
12/01/2024 18:08:56.338	SendImmediate	   at System.IO.Ports.SerialPort.Write(Byte[] buffer, Int32 offset, Int32 count)
12/01/2024 18:08:56.338	SendImmediate	   at LaserGRBL.ComWrapper.UsbSerial.Write(Byte b) in E:\LaserGRBL\LaserGRBL\ComWrapper\UsbSerial.cs:line 126
12/01/2024 18:08:56.338	SendImmediate	   at LaserGRBL.GrblCore.SendImmediate(Byte b, Boolean mute) in E:\LaserGRBL\LaserGRBL\Core\GrblCore.cs:line 1516
12/01/2024 18:08:57.440	SendImmediate	System.TimeoutException: The write timed out.
12/01/2024 18:08:57.440	SendImmediate	   at System.IO.Ports.SerialStream.Write(Byte[] array, Int32 offset, Int32 count, Int32 timeout)
12/01/2024 18:08:57.440	SendImmediate	   at System.IO.Ports.SerialPort.Write(Byte[] buffer, Int32 offset, Int32 count)
12/01/2024 18:08:57.440	SendImmediate	   at LaserGRBL.ComWrapper.UsbSerial.Write(Byte b) in E:\LaserGRBL\LaserGRBL\ComWrapper\UsbSerial.cs:line 126
12/01/2024 18:08:57.440	SendImmediate	   at LaserGRBL.GrblCore.SendImmediate(Byte b, Boolean mute) in E:\LaserGRBL\LaserGRBL\Core\GrblCore.cs:line 1516
12/01/2024 18:08:57.950	CloseCom    	Close COM3 [USER]
12/01/2024 18:08:57.951	SendImmediate	System.TimeoutException: The write timed out.
12/01/2024 18:08:57.951	SendImmediate	   at System.IO.Ports.SerialStream.Write(Byte[] array, Int32 offset, Int32 count, Int32 timeout)
[... 114 lines cut ...]
12/01/2024 18:47:50.277	CloseCom    	Close COM3 [CORE]
12/01/2024 18:47:50.279	SetStatus   	Machine status [Disconnected]
12/01/2024 18:49:23.213	Program     	---------------- PROGRAM STOP -----------------

13/01/2024 20:52:09.978	Program     	------- LaserGRBL v5.7.0 [64bit] START -------
[... cut ...]
13/01/2024 21:04:55.820	CloseCom    	Close COM8 [CORE]
13/01/2024 21:04:55.876	SetStatus   	Machine status [Disconnected]
13/01/2024 21:06:23.225	SetStatus   	Machine status [Connecting]
13/01/2024 21:06:23.227	OpenCom     	Open COM8 @ 115200 baud  (UsbSerial)
13/01/2024 21:06:23.230	SendImmediate	Send Immediate Command [0x18]
13/01/2024 21:06:33.228	OpenCom     	Connection timeout!
13/01/2024 21:06:33.230	CloseCom    	Close COM8 [CORE]
13/01/2024 21:06:33.233	SetStatus   	Machine status [Disconnected]

arkypita, 2024-01-14:
Could you please upload here the comlog.txt file that it was created with the last test?
Drag-drop it in GitHub web page, do not reply to the email because it doesn't work

phischy, 2024-01-14:
[comlog.txt](https://github.com/arkypita/LaserGRBL/files/13932798/comlog.txt)

arkypita, 2024-01-16:
Given the log you sent, look like the machine doesn't respond at all to LaserGRBL inputs.
This not depends on LaserGRBL version, or at least it is very very strange if it does.

I suggest to check LaserGRBL settings soft reset and hard reset on connect, try switching them on.

[image]

eliuth1686, 2024-02-03:
buenas tardes mi lasser funcionaba correctamente pero después ya no quiso conectarse ya intente los puertos com y sigue sin funcionar esto es lo que esta en el historial

02/02/2024 17:20:58.449	SetStatus   	Machine status [Connecting]
02/02/2024 17:20:58.451	OpenCom     	Open COM5 @ 115200 baud DTR, RTS, Ctrl-X (UsbSerial)
02/02/2024 17:20:58.456	SendImmediate	Send Immediate Command [0x18]
02/02/2024 17:20:58.794	SetStatus   	Machine status [Disconnected]
02/02/2024 17:22:31.191	SetStatus   	Machine status [Connecting]
02/02/2024 17:22:31.193	OpenCom     	Open COM5 @ 115200 baud DTR, RTS, Ctrl-X (UsbSerial)
02/02/2024 17:22:31.197	SendImmediate	Send Immediate Command [0x18]
02/02/2024 17:22:31.446	VersionInfo 	Detected Grbl v1.1h
02/02/2024 17:22:31.527	SetStatus   	Machine status [Disconnected]
02/02/2024 17:38:13.580	SetStatus   	Machine status [Connecting]
02/02/2024 17:38:13.582	OpenCom     	Open COM5 @ 115200 baud DTR, RTS, Ctrl-X (UsbSerial)
02/02/2024 17:38:13.585	SendImmediate	Send Immediate Command [0x18]
02/02/2024 17:38:13.838	SendImmediate	System.IO.IOException: Uno de los dispositivos conectados al sistema no funciona.
02/02/2024 17:38:13.838	SendImmediate	
02/02/2024 17:38:13.838	SendImmediate	   en System.IO.Ports.InternalResources.WinIOError(Int32 errorCode, String str)
02/02/2024 17:38:13.838	SendImmediate	   en System.IO.Ports.SerialStream.EndWrite(IAsyncResult asyncResult)
02/02/2024 17:38:13.838	SendImmediate	   en System.IO.Ports.SerialStream.Write(Byte[] array, Int32 offset, Int32 count, Int32 timeout)
02/02/2024 17:38:13.838	SendImmediate	   en System.IO.Ports.SerialPort.Write(Byte[] buffer, Int32 offset, Int32 count)
02/02/2024 17:38:13.838	SendImmediate	   en LaserGRBL.ComWrapper.UsbSerial.Write(Byte b) en E:\LaserGRBL\LaserGRBL\ComWrapper\UsbSerial.cs:línea 126
02/02/2024 17:38:13.838	SendImmediate	   en LaserGRBL.GrblCore.SendImmediate(Byte b, Boolean mute) en E:\LaserGRBL\LaserGRBL\Core\GrblCore.cs:línea 1516
02/02/2024 17:38:13.901	SetStatus   	Machine status [Disconnected]
02/02/2024 17:40:10.840	SetStatus   	Machine status [Connecting]
02/02/2024 17:40:10.841	OpenCom     	Open COM5 @ 115200 baud DTR, RTS, Ctrl-X (UsbSerial)
02/02/2024 17:40:10.845	SendImmediate	Send Immediate Command [0x18]
02/02/2024 17:40:11.148	SendImmediate	System.IO.IOException: Se ha especificado un dispositivo que no existe.
02/02/2024 17:40:11.148	SendImmediate	
02/02/2024 17:40:11.148	SendImmediate	   en System.IO.Ports.InternalResources.WinIOError(Int32 errorCode, String str)
02/02/2024 17:40:11.148	SendImmediate	   en System.IO.Ports.SerialStream.EndWrite(IAsyncResult asyncResult)
02/02/2024 17:40:11.148	SendImmediate	   en System.IO.Ports.SerialStream.Write(Byte[] array, Int32 offset, Int32 count, Int32 timeout)
02/02/2024 17:40:11.148	SendImmediate	   en System.IO.Ports.SerialPort.Write(Byte[] buffer, Int32 offset, Int32 count)
02/02/2024 17:40:11.148	SendImmediate	   en LaserGRBL.ComWrapper.UsbSerial.Write(Byte b) en E:\LaserGRBL\LaserGRBL\ComWrapper\UsbSerial.cs:línea 126
02/02/2024 17:40:11.148	SendImmediate	   en LaserGRBL.GrblCore.SendImmediate(Byte b, Boolean mute) en E:\LaserGRBL\LaserGRBL\Core\GrblCore.cs:línea 1516
02/02/2024 17:40:11.169	SetStatus   	Machine status [Disconnected]
02/02/2024 18:27:01.919	Issue detector	System sleep/hibernation detected (85s)

arkypita, 2024-02-03:
@eliuth1686 this could be a cable issue or a PC issue, could you please try with a different PC?

eliuth1686, 2024-02-03:
Ya lo intente con cables diferentes y con PC diferente y sigue igual sin conectarse

Obtener Outlook para Android<https://aka.ms/AAb9ysg>

MinPD291, 2024-02-13:
hi i have the same issue and tried the hard and soft reset but issue still presist. help :(

Akaisikjh, 2024-02-20:
[image]

добрый вечер у меня такая проблем не как не реагирует

Akaisikjh, 2024-02-20:
[image]

arkypita, 2024-02-21:
@Akaisikjh your machine has issue, continuously reset itself.

Akaisikjh, 2024-02-24:
[image]
Ошибка при подключении что может быть??::

Akaisikjh, 2024-02-24:
лазер подключен горить лампочка, но не как не реагирует на команды

Akaisikjh, 2024-02-25:
[image]

Akaisikjh, 2024-02-25:
[image]
у меня почему то конфигурация GRBL пустой

10ANT, 2024-03-14:
I found the solution: more than likely emergency button is clicked turn it clockwise or just try to unlock it.
[image]

Pelendrin, 2024-11-21:
Hola, la última vez que utilicé LaserGRBL se conectó sin ningún problema. Hoy no consigo que se conecte. la máquina hace un ruido de conexión como hacía habitualmente, pero el programa continúa sin conexión. Para descartar problemas del pc he probado con Universal GCode Sender, y se conecta perfectamente. También funciona con LaserWeb.
Este fragmento es de a última conexión correcta y la de hoy:

30/10/2024 23:32:42.863	Program     	------- LaserGRBL v7.12.0 [64bit] START -------
30/10/2024 23:32:43.643	OpenGL      	AUTO->FBO OpenGL 3.1.0 - Build 9.17.10.4459, Intel, Intel(R) HD Graphics 3000
30/10/2024 23:33:19.070	SetStatus   	Machine status [Connecting]
30/10/2024 23:33:19.082	OpenCom     	Open COM5 @ 115200 baud Ctrl-X (UsbSerial)
30/10/2024 23:33:19.137	SendImmediate	Send Immediate Command [0x18]
30/10/2024 23:33:19.773	VersionInfo 	Detected Grbl v1.1h
30/10/2024 23:33:19.914	SetStatus   	Machine status [Idle]
30/10/2024 23:33:20.432	Refresh Config	Refreshing grbl configuration [OnConnect]
30/10/2024 23:33:20.492	Refresh Config	Configuration successfully received! (Expected: 34 Found: 34)
30/10/2024 23:33:20.510	EnlargeBuffer	Buffer size changed to 128 [True]
30/10/2024 23:35:13.896	SetStatus   	Machine status [Jog]
30/10/2024 23:35:21.896	SetStatus   	Machine status [Idle]
30/10/2024 23:35:22.897	SetStatus   	Machine status [Jog]
30/10/2024 23:35:24.396	SetStatus   	Machine status [Idle]
30/10/2024 23:35:24.898	SetStatus   	Machine status [Jog]
30/10/2024 23:35:26.396	SetStatus   	Machine status [Idle]
30/10/2024 23:35:26.645	SetStatus   	Machine status [Jog]
30/10/2024 23:35:28.146	SetStatus   	Machine status [Idle]
30/10/2024 23:35:46.147	SetStatus   	Machine status [Jog]
30/10/2024 23:35:47.647	SetStatus   	Machine status [Idle]
30/10/2024 23:36:04.646	SetStatus   	Machine status [Jog]
30/10/2024 23:36:06.148	SetStatus   	Machine status [Idle]
30/10/2024 23:36:07.897	SetStatus   	Machine status [Jog]
30/10/2024 23:36:20.647	SetStatus   	Machine status [Idle]
30/10/2024 23:36:21.397	SetStatus   	Machine status [Jog]
30/10/2024 23:36:26.646	SetStatus   	Machine status [Idle]
30/10/2024 23:36:44.145	SetStatus   	Machine status [Jog]
30/10/2024 23:36:45.646	SetStatus   	Machine status [Idle]
30/10/2024 23:42:53.909	SetStatus   	Machine status [Jog]
30/10/2024 23:42:55.401	SetStatus   	Machine status [Idle]
30/10/2024 23:42:57.647	SetStatus   	Machine status [Jog]
30/10/2024 23:43:10.395	SetStatus   	Machine status [Idle]
30/10/2024 23:43:10.645	SetStatus   	Machine status [Jog]
30/10/2024 23:43:12.146	SetStatus   	Machine status [Idle]
30/10/2024 23:43:15.147	SetStatus   	Machine status [Jog]
30/10/2024 23:43:18.397	SetStatus   	Machine status [Idle]
30/10/2024 23:43:18.647	SetStatus   	Machine status [Jog]
30/10/2024 23:43:23.147	SetStatus   	Machine status [Idle]
30/10/2024 23:43:24.897	SetStatus   	Machine status [Jog]
30/10/2024 23:43:26.396	SetStatus   	Machine status [Idle]
30/10/2024 23:43:26.647	SetStatus   	Machine status [Jog]
30/10/2024 23:43:34.646	SetStatus   	Machine status [Idle]
30/10/2024 23:43:35.645	SetStatus   	Machine status [Jog]
30/10/2024 23:43:37.148	SetStatus   	Machine status [Idle]
30/10/2024 23:44:29.398	SetStatus   	Machine status [Jog]
30/10/2024 23:44:30.896	SetStatus   	Machine status [Idle]
30/10/2024 23:44:33.397	SetStatus   	Machine status [Jog]
30/10/2024 23:44:34.646	SetStatus   	Machine status [Idle]
30/10/2024 23:44:37.398	SetStatus   	Machine status [Jog]
30/10/2024 23:44:38.897	SetStatus   	Machine status [Idle]
30/10/2024 23:44:41.396	SetStatus   	Machine status [Jog]
30/10/2024 23:44:51.896	SetStatus   	Machine status [Idle]
30/10/2024 23:44:53.648	SetStatus   	Machine status [Jog]
30/10/2024 23:44:57.645	SetStatus   	Machine status [Idle]
30/10/2024 23:45:08.395	SetStatus   	Machine status [Jog]
30/10/2024 23:45:11.146	SetStatus   	Machine status [Idle]
30/10/2024 23:45:11.397	SetStatus   	Machine status [Jog]
30/10/2024 23:45:12.646	SetStatus   	Machine status [Idle]
30/10/2024 23:45:34.398	SetStatus   	Machine status [Jog]
30/10/2024 23:45:37.396	SetStatus   	Machine status [Idle]
30/10/2024 23:45:38.648	SetStatus   	Machine status [Jog]
30/10/2024 23:45:39.896	SetStatus   	Machine status [Idle]
30/10/2024 23:45:40.147	SetStatus   	Machine status [Jog]
30/10/2024 23:45:41.647	SetStatus   	Machine status [Idle]
30/10/2024 23:45:46.396	SetStatus   	Machine status [Jog]
30/10/2024 23:45:47.895	SetStatus   	Machine status [Idle]
30/10/2024 23:47:38.578	OpenFile    	Open C:\Users\Galegarte\Documents\GALEGARTE\halloween\Hucha.png
[... 56 lines cut ...]
21/11/2024 17:30:21.952	SetStatus   	Machine status [Disconnected]
21/11/2024 17:30:32.044	SetStatus   	Machine status [Connecting]
21/11/2024 17:30:32.045	OpenCom     	Open COM4 @ 115200 baud Ctrl-X (UsbSerial)
21/11/2024 17:30:32.079	CloseCom    	Close COM4 [CORE]
21/11/2024 17:30:32.079	SendImmediate	Send Immediate Command [0x18]
21/11/2024 17:30:32.084	SetStatus   	Machine status [Disconnected]
21/11/2024 17:30:38.652	SetStatus   	Machine status [Connecting]
21/11/2024 17:30:38.653	OpenCom     	Open COM3 @ 115200 baud Ctrl-X (UsbSerial)
21/11/2024 17:30:38.775	SendImmediate	Send Immediate Command [0x18]
21/11/2024 17:30:48.656	OpenCom     	Connection timeout!
21/11/2024 17:30:48.656	CloseCom    	Close COM3 [CORE]
21/11/2024 17:30:48.669	SetStatus   	Machine status [Disconnected]
21/11/2024 17:31:19.976	SetStatus   	Machine status [Connecting]
[... cut ...]
21/11/2024 17:31:20.030	SendImmediate	Send Immediate Command [0x18]
21/11/2024 17:31:20.030	CloseCom    	Close COM4 [CORE]
21/11/2024 17:31:20.035	SetStatus   	Machine status [Disconnected]
21/11/2024 17:31:25.840	Program     	---------------- PROGRAM STOP -----------------

Me llama especialmente la atención estas líneas:
01/11/2024 18:04:31.448	Issue detector	System sleep/hibernation detected (10797s)
07/11/2024 17:42:46.376	Issue detector	System sleep/hibernation detected (517093s)

Muchas gracias por tu ayuda y dedicación
Un saludo

Pelendrin, 2024-11-21:
Vale. No sé si tiene algo que ver o no, pero lo he solucionado volviendo a instalar el driver CH340 (Heramientas/Instalar Driver CH340)

Gracias igualmente
Un saludo

Moskovcev, 2025-05-21:
Я полный юзер и снова о подключении. пишет ошибка, хотя первый ра все было нормально. Поменял комп, залил погу и драйвер снова.
Последние строки из Журнала
21.05.2025 08:58:19.116	OpenCom     	Open COM5 @ 4800 baud Ctrl-X (UsbSerial)
21.05.2025 08:58:19.119	SetStatus   	Machine status [Disconnected]
21.05.2025 08:58:19.126	SendImmediate	Send Immediate Command [0x18]
21.05.2025 08:58:26.745	SetStatus   	Machine status [Connecting]
21.05.2025 08:58:36.747	OpenCom     	Connection timeout!
21.05.2025 08:58:36.747	CloseCom    	Close COM5 [CORE]
21.05.2025 08:58:36.749	SetStatus   	Machine status [Disconnected]
21.05.2025 08:59:07.701	Program     	---------------- PROGRAM STOP -----------------

21.05.2025 09:05:20.214	Program     	------- LaserGRBL v7.14.0 [64bit] START -------
21.05.2025 09:05:21.152	OpenGL      	AUTO->FBO OpenGL 4.3.0, NVIDIA Corporation, GeForce GT 650M/PCIe/SSE2
21.05.2025 09:06:04.218	SetStatus   	Machine status [Connecting]
21.05.2025 09:06:04.241	OpenCom     	Open COM5 @ 4800 baud Ctrl-X (UsbSerial)
21.05.2025 09:06:04.264	SendImmediate	Send Immediate Command [0x18]
21.05.2025 09:06:14.223	OpenCom     	Connection timeout!
21.05.2025 09:06:14.223	CloseCom    	Close COM5 [CORE]
21.05.2025 09:06:14.227	FixCH340    	Detected CH340 driver:

И в списке программ у меня их три!!!
Разницы не заметил, но запускал все три.

[image]

Что делать, куда бежать??????