arkypita/LaserGRBL #2191: svg file spindle speed command contains trailing space when saved to a TF card

ojaitom, 2024-02-04:
**Describe the bug**
Saving an svg file to the computer for transfer to the laser via TF card does not work for my configuration.  Deleting the trailing space between the power value and the line feed corrects the issue and allows use of the saved gcode file

**To Reproduce**
Steps to reproduce the behavior:
1. Go to file 
2. Click on save (Advanced Options)
3. Click on Save
4. See error 

**Expected behavior**
Spindle speed command should not have any space following the power value prior to the line feed

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Hardware and software configuration (please complete the following information):**
 - Windows Version	 10 Pro 10.0.19045 Build 19045

 - Creality Falcon Module for ender 3 printers
 - Grbl version 1.3a
 - LaserGRBL version v5.7.0

**Additional context**
Add any other context about the problem here.

arkypita, 2024-02-05:
Hi, thanks for your bug report.
Even if it's not really a LaserGRBL bug (the gcode allows spaces anywhere on the line) I can make the necessary changes to fix it.
However, the problem should be addressed by the machine's firmware developers.

[image]

ojaitom, 2024-02-05:
Thanks for the quick response.  I really appreciate Lasergrbl as a great alternative to what other products are available.  I'll look in to submitting a problem report to Creality, though I don't expect as quick or thorough a response.  Keep up the good work.

Cheers,

Tom 

image.png (view on web)

"Organization", "name": "GitHub", "url": "<a href="https://github.com">https://github.com</a>" } } ]