arkypita/LaserGRBL #2205: homing machine without sensors on z axia (windows )

Bartekn86, 2024-02-18:
Hi
is it possible to add an option that eliminates the need to homing the z axis (many plotters have only 2 axes - no sensors on the z axis because it is manual)
Windows compilation does not have a file responsible for such a function, unlike Linux.
I did not find in the documentation the possibility of basing only the x and y axes and completely omitting the z axis

have a good day

MRT-77, 2024-02-19:
I solved this problem by adding a jumper in the z-axis end stop connector

arkypita, 2024-02-19:
It is the controller firmware that handle homing, so there is no option that could be added in LaserGRBL to change this. You should re-build your firmware