arkypita/LaserGRBL #2217: grblhal + teensy 4.1

distebia, 2024-03-06:
Salve, ulitizzo grblhal con teensy 4.1 (iMXRT1062) e gestisco tutto con Lightburn e connessione ethernet. tutto funziona perfettamente. Oggi installo LaserGrbl, mi connetto ethernet, leggo perfettamente il parametri del controller ma si comporta in modo strano. Naturalmente tanti parametri del firmware vengono visualizzati privi di descrizione, ma questo non è un problema, li riconosce. Eseguo $h e viene eseguito non rispettando i parametri del firmware (5000) ma eseguito a 1000 (come da impostazione in basso a sx. Inoltre il mio ciclo di homing prevede 2 ripetizioni e il programma quando incontra il finecorsa si blocca all'istante...sbaglio qualcosa o è giusto il comportamento?

arkypita, 2024-03-06:
$h (homing) viene eseguito completamente e autonomamente dalla scheda.

Se sei sicuro di aver inviato $h (ne hai il feedback nella colonna a sx dove escono i vari comandi inviati) tutto ciò che segue dipende dalla scheda, non certo dal programma che lo ha inviato.

Quindi secondo me non hai inviato $h ma un altro comando. Quale pulsante hai usato per inviare $h?

distebia, 2024-03-06:
sbagliavo io, premevo il pulsante sbagliato 
[image]

distebia, 2024-03-07:
Ma lasergrbl gestisce il clustering gcode per le incisioni a toni di grigio?

arkypita, 2024-03-07:
Ciao, è una funzione che mi stanno chiedendo di implementare (per la AlgoLaser) ma per ora non sono riuscito a trovare il tempo per farla.

distebia, 2024-03-07:
e il plugin grblhal ppi può aiutare?

arkypita, 2024-03-07:
direi di no

distebia, 2024-03-07:
ok, grazie

closed 2024-12-29 by distebia