arkypita/LaserGRBL #2219: Images are being squashed

hawkeyelaser, 2024-03-07:
**Describe the bug**
Although it is showing up properly in the preview screen, when I hit engrave, my images are being squished. going from a circle or square to a oval or rectangle 
has done it since update 6.1
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
print as shown

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware and software configuration (please complete the following information):**
 - Windows version 11
 -comgrow v1 10 watt
 version 6.2
[image]
[image]
[image]

**Additional context**
Add any other context about the problem here.

StuartB4, 2024-03-08:
Check to make sure the toothed wheels on the motors are all tight and not slipping on the motor shaft.
Check the belts are not too tight.
Check that the laser moves nice and smooth in both X and Y axis. 
Put the laser all the way to the left and then Tilt the machine up from the left to about 45 degrees and the laser should run 
nice and smooth without having to push it, then do same with Y axis. 
Check the wiring to all motors (both ends of the leads)  for loose connections.
Check the GRBL Configuration settings fo $100 and $101 (steps/mm) usually they would both be the same number.

hawkeyelaser, 2024-03-08:
Great thank you.

The $100 and $101 had changed

StuartB4, 2024-03-08:
You're welcome hawkeyelaser. Glad you sorted it out. 👍 :)