arkypita/LaserGRBL #2221: Cant use max power on my laser module

krishan4554, 2024-03-08:
hello 
Firstly thanks a lot founder of [LaserGRBL](https://github.com/arkypita/LaserGRBL)
Details of cnc
 power adapter 12v 5A
[12v 40w laser module  ](https://www.aliexpress.com/item/1005004846263876.html?spm=a2g0o.productlist.main.9.c36c1931bAnQzg&algo_pvid=8320da2a-78d6-4c97-b1bf-b3dd394f31a9&aem_p4p_detail=202403080755019048679466123440003369395&algo_exp_id=8320da2a-78d6-4c97-b1bf-b3dd394f31a9-4&pdp_npi=4%40dis%21LKR%2121380.71%2121380.71%21%21%2169.58%2169.58%21%402103253417099133013907314e145f%2112000030723917259%21sea%21LK%210%21AB&curPageLogUid=6PfwrQAO8CEN&utparam-url=scene%3Asearch%7Cquery_from%3A&search_p4p_id=202403080755019048679466123440003369395_5#nav-review
)
 2 Nema steper
 cnc v3 board 
 Arduino uno
please check the screenshot I tried changing PWM values but 160 is max engraving  i can increase 255 but the engraving is not continuously going 
[image]

My machine parameter are  
[image]

PLEASE HELP ME TO SLOVE THIS

arkypita, 2024-03-09:
Maybe you power supply is underrated, and it is not able to drive both motor and laser at its full power?

krishan4554, 2024-03-09:
Thanks for the reply 
The cnc is working with these [parameters](https://private-user-images.githubusercontent.com/162727164/311299628-729fefde-1cb5-45e2-87fd-95f64065325e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAwMDcyMjUsIm5iZiI6MTcxMDAwNjkyNSwicGF0aCI6Ii8xNjI3MjcxNjQvMzExMjk5NjI4LTcyOWZlZmRlLTFjYjUtNDVlMi04N2ZkLTk1ZjY0MDY1MzI1ZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMwOVQxNzU1MjVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jZmNmZGRiOGZlNTIwYWM4ZTEyMmM4NzVkNTE1Nzc1YTM1MGRjMDU3YzUwYjczOTFkMWU1NWNhOWFjZTJkNjM5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.JrY8zQxaaBa_apccYbu14WFlKoA_gIry5G_ZIgZFCcU) 
how can i check the power  issue  please